hifiberry/hifiberry-os #117: No Password for main UI.

neilt, 2020-07-09:
**Describe the bug**
There is no password for access the main UI.

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD
The "About" screen does NOT appear when clicking.

**HiFiBerry sound card**
Dac+ Pro

**To Reproduce**
1. Navigate browser to http://wholehouseaudio.local
2. UI comes up with no password request.

**Expected behavior**
Expected to have to enter a password.

**Screenshots**
No applicable

**Browser (if applicable)**
Safari Version 13.1.1 (15609.2.9.1.2)
macOS Catalina 10.15.5 (19F101)

**Additional context**
None

hifiberry, 2020-07-09:
The software is to be used in your protected home environment. There is no user management implemented at all. This is how it is expected to work.

closed 2020-07-09 by hifiberry

neilt, 2020-07-09:
There is no such thing as a protected home environment.  I can hack most WiFi in under 20 minutes.  I don't need any user management just a UI password, that's all.  Plus most users have hacked PCs setting on the network.  Again no protected home environment.  This pretty much makes HiFiBerryOS untrustable.

hifiberry, 2020-07-09:
Feel free to implement this and send a pull request:
https://github.com/hifiberry/hifiberry-os/blob/master/CONTRIBUTING.md

neilt, 2020-07-09:
Sure thing, I'll get right on that.

hifiberry, 2020-07-09:
Just make sure that it's optional as most users do not want a password for their music player at home.