hifiberry/hifiberry-os #118: Software updates "regular" do not work well

keesma, 2020-07-10:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
HifiberryOS cannot download previous software versions. it looks like they are not available online.

**HiFiBerryOS version**
20204020

**HiFiBerry sound card**
HifiBerry Beocreate.

**To Reproduce**
Steps to reproduce the behavior:
1. Set software update to regular (not that there is a recent update 20200708)
2. Start manual update (the GUI has the same behaviour but not a clear error message)

2. See error below: "Could not download updater https://www.hifiberry.com/dev/updater-20200527-pi3.tar.gz, aborting"

**Expected behavior**
Update hifiberryos to 20200527.

**Screenshots**
# /opt/hifiberry/bin/update
updating from 20200424 to 20200527...
downloading https://www.hifiberry.com/dev/updater-20200527-pi3.tar.gz...
curl: (22) The requested URL returned error: 404                                                                                                                    
Could not download updater https://www.hifiberry.com/dev/updater-20200527-pi3.tar.gz, aborting

**Browser (if applicable)**
 - not relevant

**Additional context**
If you set automatic updates to 'Quick' the most recent software version is downloaded and installed..
This is the first time an update is installed.

hifiberry, 2020-07-10:
Looks like we cleaned some older stuff to early. Please try again now.

keesma, 2020-07-11:
I still get the same issue on the command line:
Could not download updater https://www.hifiberry.com/dev/updater-20200527-pi3.tar.gz, aborting

hifiberry, 2020-07-14:
I'm sorry for this. I have now updated the latest release to 20200708 which seems to work quite well. Please try again.

closed 2020-07-14 by keesma