hifiberry/hifiberry-os #119: DAC+ADC Pro board not recognized automatically upon installation of the OS on Pi4

Boobybandit, 2020-07-14:
> (this is actually 2 bug reports in 1, but did not know how to separate it without seeming like the same thing).

Hardware: Installed the DAC+ADC pro on the Pi4 and put the aamp60 on top of it for the output to my speakers. 
Software: Installed the latest HifiberryOS v20200424.

Issue 1:
Everything works just fine (spotify, radio), except for the analog input not showing up as a source.

After emailing my debug-rapport to Daniel from support he said the following:

> > I see that the hardware isn't configured correctly. This looks like a bug in HiFiBerryOS. You might fix this by hand as follows:
> 
> - login via SSH
> - mount -o remount,rw /boot
> - edit /boot.config.txt and replace "hifiberry-dacplus by hifiberry-dacplusadcpro

So I did. After a reboot the interface showed the analogue input option finally!

Issue 2: 
After following the steps as described above here. Still no sound coming out of my speakers when selecting and connecting the analogue input. Should still be a software issue as the input did work on volumio and  I installed the same DAC+ADC + aamp60 on my old Pi2 and there the analogue input did provide sound! 

Do not know if it matters but I gave the "mount -o remount,rw /boot" via Putty and then edited the file via Winscp.  

[hifiberry-debug (1).zip](https://github.com/hifiberry/hifiberry-os/files/4919384/hifiberry-debug.1.zip)

hifiberry, 2020-07-14:
Thanks for reporting this. Looks like a combination of several things that are not working perfectly - an old config.txt workaround and missing ADC detection.

hifiberry, 2020-07-14:
Fixed with https://github.com/hifiberry/hifiberry-os/commit/a289579ac0c1df35eb241b87fd1cb10e1a0d31b9

Boobybandit, 2020-07-14:
Wow that's quick! Is there any way for me to update my current image with the adjustments you made?

hifiberry, 2020-07-15:
No, as this requires a new build. However, as your system is working now, I would recommend to leave it as it is.

Boobybandit, 2020-07-15:
Hi, so actually, issue 2 is still ongoing for me. I don't get any sound and don't know what the problem is. Do you think issue 2 is caused by the manual fix I did for issue 1 and will also be solved by a289579?

At this moment the DAC+ADC only works on my PI2, but limited as that build does not support Spotify and is somewhat slow due to the cpu of the PI2.

hifiberry, 2020-07-15:
I see that you merged 2 bug reports into one :(

Probably the second has nothing to do with the first.

Boobybandit, 2020-07-15:
> I see that you merged 2 bug reports into one :(
> 
> Probably the second has nothing to do with the first.

Yes, fixed that, I opened a new report.

Boobybandit, 2020-08-11:
I just did a clean install of the newest software (image on the website with date 08-05) and it works out of the box! My other issue https://github.com/hifiberry/hifiberry-os/issues/121 is also fixed, so everything seems to be working perfectly now. Just a slight crackling when using the analog input, but I'm not even sure.

Very happy for now as the automatic switching of inputs was the most important feature for me to stick with HifiberryOS! 

Thanks, thread can be closed and marked as fixed.

closed 2020-09-09 by hifiberry