hifiberry/hifiberry-os #120: 20200706 / 20200708 BT issues on RPi B+

goda17, 2020-07-14:
**Describe the bug**
Bluetooth on a RPi 3B+ will no longer enable; now behaves exactly like a 3B starting with 20200706. Works as expected on 3A+. Last worked fine on 20200527. Reimaging to 20200527 makes it work again.  

**HiFiBerryOS version**
20200706

**HiFiBerry sound card**
Amp+

**To Reproduce**
Steps to reproduce the behavior:
1. Update to 20200706 on a RPi 3B+
2. Go to Sources > Bluetooth; it will be unchecked
3. Check to enable it; refresh page after a bit
4. It will be disabled; no previously paired devices can connect or pair

**Expected behavior**
BT should enable and be available for connections / pairing

**Screenshots**
[image]

**Browser (if applicable)**
 n/a

**Additional context**
# cat /etc/hifiberry.state
CURRENT_PIMODEL=3B+
CURRENT_MIXER_CONTROL="Softvol"
CURRENT_CARD="Amp+"
CURRENT_HW="card 0"
CURRENT_HW_SHORT=0
CURRENT_SYSNAME="Office"
CURRENT_PLAYERS=""
CURRENT_HATINFO=""
CURRENT_FEATURES="pi bluetooth pi3orlater arm7 localui"
PIVERSION=3

hifiberry, 2020-07-15:
Unfortunately I don't have a Pi3B+ for a test here and on the Pi4 it's working fine. 

We need some tests. Please run the following commands on the 202007 release and post the output:

systemctl enable --now bluetooth.service bluealsa.service bluealsa-aplay.service
systemctl status bluetooth.service bluealsa.service bluealsa-aplay.service

goda17, 2020-07-16:
**Here's the output from a 3B+:**

● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor
preset: enabled)
     Active: inactive (dead)
  Condition: start [[0;1;38;5;185mcondition failed[[0m at Wed 2020-07-15 21:05:3
3 EDT; 5s ago
             └─ ConditionPathIsDirectory=/sys/class/bluetooth was not met
       Docs: man:bluetoothd(8)

Jul 15 21:05:33 office systemd[1]: Condition check resulted in Bluetooth service
 being skipped.
Jul 15 21:05:33 office systemd[1]: Condition check resulted in Bluetooth service
 being skipped.
Jul 15 21:05:33 office systemd[1]: Condition check resulted in Bluetooth service
 being skipped.

[[0;1;32m●[[0m bluealsa.service - BlueALSA
     Loaded: loaded (/usr/lib/systemd/system/bluealsa.service; enabled; vendor p
reset: enabled)
     Active: [[0;1;32mactive (running)[[0m since Fri 2020-04-17 08:37:15 EDT; 2
months 28 days ago
   Main PID: 225 (bluealsa)
      Tasks: 3 (limit: 1547)
standard input

hifiberry, 2020-07-16:
Ok, it looks like some problem with the update as a fresh installation works here. Can you try a fresh installation of the latest releas to see if this works on your system:
https://github.com/hifiberry/hifiberry-os/releases/tag/v20200708

hifiberry, 2020-07-21:
@goda17 : Any feedback on a fresh installation?

fjfricke, 2020-07-22:
On my Raspberry Pi 4, bluetooth discovery doesn't work on a fresh install. Might this have to do anything with this bug?

steinmb, 2020-08-28:
Upgraded to 20200805 and also lost BT on a 3+

systemctl status bluetooth.service bluealsa.service bluealsa-aplay.service

```
[[0;1;32m●[[0m bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: [[0;1;32mactive (running)[[0m since Fri 2020-08-28 09:24:27 CEST; 2min 47s ago
       Docs: man:bluetoothd(8)
   Main PID: 13284 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 1547)
     Memory: 1.1M
     CGroup: /system.slice/bluetooth.service
             └─13284 /usr/libexec/bluetooth/bluetoothd

Aug 28 09:24:27 kontor systemd[1]: Starting Bluetooth service...
Aug 28 09:24:27 kontor bluetoothd[13284]: Bluetooth daemon 5.54
Aug 28 09:24:27 kontor systemd[1]: Started Bluetooth service.
Aug 28 09:24:27 kontor bluetoothd[13284]: Starting SDP server
Aug 28 09:24:27 kontor bluetoothd[13284]: Bluetooth management interface 1.14 initialized

[[0;1;32m●[[0m bluealsa.service - BlueALSA
     Loaded: loaded (/usr/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
     Active: [[0;1;32mactive (running)[[0m since Thu 2020-08-27 16:20:17 CEST; 17h ago
   Main PID: 230 (bluealsa)
      Tasks: 3 (limit: 1547)
     Memory: 3.8M
     CGroup: /system.slice/bluealsa.service
             └─230 /usr/bin/bluealsa -i hci0 -p a2dp-sink --a2dp-volume

Aug 27 16:20:17 kontor systemd[1]: Starting BlueALSA...
Aug 27 16:20:17 kontor systemd[1]: Started BlueALSA.
Aug 27 16:20:53 kontor bluealsa[230]: /usr/bin/bluealsa: W: Couldn't get managed objects: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.bluez': timed out (service_start_timeout=25000ms)

[[0;1;32m●[[0m bluealsa-aplay.service - BlueALSA aplay
     Loaded: loaded (/usr/lib/systemd/system/bluealsa-aplay.service; enabled; vendor preset: enabled)
     Active: [[0;1;32mactive (running)[[0m since Thu 2020-08-27 16:20:19 CEST; 17h ago
   Main PID: 324 (bluealsa-aplay-)
      Tasks: 2 (limit: 1547)
     Memory: 956.0K
     CGroup: /system.slice/bluealsa-aplay.service
             ├─324 /bin/sh /opt/btspeaker/bluealsa-aplay-start
             └─328 /usr/bin/bluealsa-aplay --pcm-buffer-time=250000 00:00:00:00:00:00 --volume-control Digital

Aug 27 16:20:17 kontor systemd[1]: Starting BlueALSA aplay...
Aug 27 16:20:19 kontor systemd[1]: Started BlueALSA aplay.
Aug 27 16:20:30 kontor bluealsa-aplay-start[328]: /usr/bin/bluealsa-aplay: W: Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files
```

/opt/hifiberry/bin/debuginfo

```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   621  100   621    0     0  27000      0 --:--:-- --:--:-- --:--:-- 28227
cat: /proc/asound/card1/pcm0p/sub0/hw_params: No such file or directory
cat: /proc/asound/card2/pcm0p/sub0/hw_params: No such file or directory
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3351  100  3351    0     0   409k      0 --:--:-- --:--:-- --:--:--  467k
cat: /etc/mopidy.conf: No such file or directory
usage: dsptoolkit [-h] [--delay DELAY] [--host HOST] [--timeout TIMEOUT]
                  {apply-fir-filter-left,apply-fir-filter-right,apply-fir-filters,apply-iir-filters,apply-iir-filters-left,apply-iir-filters-right,apply-rew-filters,apply-rew-filters-left,apply-rew-filters-right,apply-settings,check-eeprom,clear-iir-filters,get-checksum,get-limit,get-loudness,get-memory,get-meta,get-prog,get-samplerate,get-volume,get-xml,install-profile,load,loop-read-dec,loop-read-hex,loop-read-int,loop-read-reg,mute,read-dec,read-hex,read-int,read-reg,reset,save,servers,set-limit,set-loudness,set-volume,store,store-filters,store-settings,tone-control,unmute,version,write-mem,write-reg}
                  [parameters [parameters ...]]
dsptoolkit: error: the following arguments are required: command, parameters
 Performance counter stats for 'system wide':

          40042.07 msec cpu-clock                 #    4.000 CPUs utilized
             55269      context-switches          #    0.001 M/sec
               459      cpu-migrations            #    0.011 K/sec
               253      page-faults               #    0.006 K/sec
        3024029528      cycles                    #    0.076 GHz
        1311593077      instructions              #    0.43  insn per cycle
         147813953      branches                  #    3.691 M/sec
          23971545      branch-misses             #   16.22% of all branches

      10.010896707 seconds time elapsed
```

/opt/hifiberry/bin/check-system

```
HiFiBerryOS version: 20200805
Pi model: 3B+
Sound card: DAC+/Amp2
Sound card ID: 0
Mixer: Digital
Features: pi bluetooth pi3orlater arm7 localui
Linux version: Linux kontor 4.19.127-v7 #1 SMP Tue Aug 4 16:47:49 UTC 2020 armv7l GNU/Linux
/data mounted: OK
BT devices:
WiFi devices: wlan0
audiocontrol2: running
beocreate2: running
bluetooth: running
bluealsa: running
bluealsa-aplay: running
dlnampris: running
mpd: running
mpd-mpris: running
pigpiod: running
raat: not running
shairport-sync: running
sigmatcp: running
snapcastmpris: not running
spotify: running
squeezelite: not running
sshd: running
ympd: running
weston: not running
cog: not running
```

hifiberry, 2020-08-28:
What do you mean by "lost"? according to your logs, everything is fine.

steinmb, 2020-08-28:
Lost as in no longer working. Show up in the device list though no longer able to connect to it. When I dropped the pairing I was no longer able to pair it. Do not even broadcast. Do not show up as a pairable device.

hifiberry, 2020-08-31:
Try the following:

rm /etc/hifiberry.state
/opt/hifiberry/bin/reconfigure-players
reboot

steinmb, 2020-08-31:
Yay! That did it. BT now works as normal. Thank you so much.

```
# rm /etc/hifiberry.state

# /opt/hifiberry/bin/reconfigure-players
/etc/hifiberry ~
~
Sound card settings have been changed, reconfiguring...
Applying settings:
 Players:  (was )
 Raspberry Pi model: 3B+
 System name: Kontor (was )
 Sound card: DAC+/Amp2 (was )
 HW device: card 0 (was )
 ALSA mixer control: Digital (was )
 Recording:  (was )
 Vendor/product HiFiBerry DAC+ (was )
 Features: pi bluetooth pi3orlater arm7 localui (was )

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
 updated mixer
 updated system name
Restarting spotify
Reconfiguring Shairport-sync
 updated mixer
 updated system name
Restarting shairport-sync
Reconfiguring Bluetooth
 updated system name
 updated mixer
Restarting bluetooth.service
Reconfiguring squeezelite
 updated system name
Reconfiguring MPD
Restarting mpd
 updated mixer
Reconfiguring DLNA
Reconfiguring Mopidy
modidy not installed, ignoring
sed: /etc/mopidy.conf: No such file or directory
 updated mixer
Reconfigure AudioControl2
Restarting audiocontrol2
 updated mixer
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
 updated card type
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'sound'.
 updated mixer
["pi","bluetooth","pi3orlater","arm7","localui"]
Settings updated for 'system'.
Restarting beocreate2
 restarting Beocreate

# reboot
```

hifiberry, 2020-08-31:
Ok, this will be improved in the next release.

hifiberry, 2020-09-09:
This should be fixed now in 20200908

closed 2020-09-09 by hifiberry