hifiberry/hifiberry-os #121: No sound output when selecting analogue input on RPI4 with DAC+ADC Pro and aamp60

Boobybandit, 2020-07-15:
**Describe the bug**
On initial installation my DAC+ADC Pro was not recognized as such, and therefore not displaying the analogue input option. After applying the manual fix described by me in https://github.com/hifiberry/hifiberry-os/issues/119#issue-656668454 I do have the analogue input option, however, this still does not provide any sound output. I attached the debug file.

**HiFiBerryOS version**
v20200424 running on a RPI4

**HiFiBerry sound card**
DAC+ADC pro + aamp60

**Additional context**
I tested the same hardware (DAC+ADC Pro + aamp60) on a RPI2 HifiberryOS installation and there it did provide sound output from the analogue input. I tried several fresh installs on different hardware combinations (I ordered the above combination twice), without any luck. So it would seem like a software problem with the HifiberryOS for RPI4. 

[hifiberry-debug.1.zip](https://github.com/hifiberry/hifiberry-os/files/4923621/hifiberry-debug.1.zip)

Boobybandit, 2020-07-20:
Just a reminder regarding this issue.

hifiberry, 2020-07-20:
No need for reminders. The issue is open, but we're very busy with other stuff at the moment.

Boobybandit, 2020-07-20:
Hi, apologies. Wasn't sure if you guys saw this one, hence the reminder. 

I do hope the issue will be fixed soon though as I specifically bought the DAC+ADC Pro + aamp60 instead of just the AMP2. 

I'll be on standby if you guys need more info!

hifiberry, 2020-07-22:
I might have an idea what's happening here. Please login via SSH and post the output of 

 ls /etc/features

Boobybandit, 2020-07-22:
Hi,

This is the output.

analoginput  arm7  bluetooth  localui  pi  pi3orlater

Do note that I applied the manual solution from 656668454

hifiberry, 2020-07-23:
That's fine. I noticed another thing that might be different on the Pi4. Please try the following:

systemctl disable pigpiod
reboot

hifiberry, 2020-07-24:
Ok, might still be some other issue. I have tested updated pigpio settings with a new release. You might try this (this is basically the pre-release of the upcoming July release) and try this by

/opt/hifiberry/bin/update --experimental

Boobybandit, 2020-07-24:
Hi, so that worked! Don't have a lot of time right now, but wanted to let you know that sound came out of the speaker finally 👍

hifiberry, 2020-07-24:
Good to hear. We'll do some final tests and if everything works well for some day, this will be finally released next week.

Boobybandit, 2020-07-24:
So I tested it a little and seems to work fine. The first 5 minutes or so there seemed to be like a microstutter. Maybe buffering or something. After that it worked just fine. 

As this is the first time that I can really experiment with the analog output there are some things that I encounter. Like when the analog input is active, other inputs (Spotify connect) cannot be started. Even when unplugging the cable Spotify won't start. Manually switching the input does not work either. 

I think it is best to close this issue. I'll take the weekend to test and then open a "analog input issuess" report next week. Some of the things that I run into may be like this by design, others may be actual RPI (4) bugs.

hifiberry, 2020-07-24:
If you have enabled auto-start on analog input, it will always override other players if a signal is detected. Especially when using "high" input sensitivity even the noise of an open input might be enough to trigger this.

Boobybandit, 2020-07-26:
Hi, yes, that seems logical. 

Today I ran some extensive tests and can report the following.

- Did a clean install with the latest image from the website and them immediately updated to the experimental version as provided. 
- Analog input did not show, only after applying the manual fix of editing the config file from dacplus to dacplusadcpro. 
- After this, analog input works! However, that's all.

Somehow when the analog input is enabled, it blocks everything else.
- This occurs both when using the regular enabled option and also when using "start when signal detected". 
- It does not matter whether I manually remove the plug from the DAC/ADC board. Even with no jack plugged in, the analog input remains active. 
- The only solution for me to play Spotify again is to disable the analog input option in the sources menu. 
- So recap: I cannot leave the analog input enabled in the sources menu and then select another source in the "now playing" section. The analog input persists, even when nothing is connected.

So at this moment it is workable by manually toggling the analog input. And even then sometimes random bugs happen. 

Would you like me to make a new report for this issue?

- Auto-start on analog input does not work (start when signal detected)
-

Boobybandit, 2020-08-01:
Hi would you like me to make a new report for the bug in my last message or should we continue this thread?

hifiberry, 2020-08-04:
You can create a new bug report for the "stop" bug. However, we can't see this behaviour here. Therefore, we might not be able to fix it.

closed 2020-08-04 by hifiberry