hifiberry/hifiberry-os #123: write accesses to the file system

soundart, 2020-07-20:
**Describe the bug**

Some packages seem to write to the filesystem which will reduce the life time of the sd-cards.

**HiFiBerryOS version**

20200720

**HiFiBerry sound card**

DAC+ Pro

**To Reproduce**

ssh into the device, list /var/log

```
$ ssh root@192.168.168.227 ls -alrt /var/log
total 32
-rw-r--r--  1 root root            1040 Apr 17 14:37 README
drwx------  2 root root            4096 Apr 17 14:37 private
-rw-rw-r--  1 root utmp               0 Apr 17 14:37 lastlog
-rw-rw----  1 root utmp               0 Apr 17 14:37 btmp
drwxr-xr-x 11 root root            4096 Apr 17 14:37 ..
-rw-r--r--  1 root root             165 Apr 17 14:37 hifiberry.log
drwxr-xr-x  5 root root            4096 Apr 17 14:37 .
drwxr-sr-x  3 root systemd-journal 4096 Jul 20 18:37 journal
drwxr-xr-x  3 root root            4096 Jul 20 18:42 samba
-rw-rw-r--  1 root utmp            1920 Jul 20 18:55 wtmp

```
systemd, samba and busybox(wtmp) seem to use that directory. 

**Expected behavior**

no writes to flash based filesystems. Maybe provide tmpfs for /var/log as fallback if a package is misconfigured?

**Screenshots**

**Browser (if applicable)**

**Additional context**

this happend using git branch development fa17e561354252dcc29698616730bee6e3ef902d

I think it is busybox at least:

```
buildroot-3/build/busybox-1.31.1$ strings busybox | grep wtmp
updwtmpx
        -w      Only write a wtmp record
/var/log/wtmp
write_wtmp
bb_path_wtmp_file
updwtmpx@@GLIBC_2.4

```
There seems to be a config option:

```
config FEATURE_UTMP
	bool "Support utmp file"
	default y
	help
	The file /var/run/utmp is used to track who is currently logged in.
	With this option on, certain applets (getty, login, telnetd etc)
	will create and delete entries there.
	"who" applet requires this option.

config FEATURE_WTMP
	bool "Support wtmp file"
	default y
	depends on FEATURE_UTMP
	help
	The file /var/run/wtmp is used to track when users have logged into
	and logged out of the system.
	With this option on, certain applets (getty, login, telnetd etc)
	will append new entries there.
	"last" applet requires this option.

```

hifiberry, 2020-07-20:
This isn't a bug, but more a feature request. In general, utmp and wtmp will rarely be written. While this might be improved, it definitely hasn't a high priority.

soundart, 2020-07-21:
Yes I agree feature-request fits.

I am a bit sensitive because I lost some larger(€€€) sd-cards initially because I misconfigured mpd logging and db files on my openwrt based system.  After some weeks uptime they were not working anymore.

hifiberry, 2020-07-21:
MPD db files will be stored on the SD card anyway. Also many other things will write to the SD card as this is necessary (e.g. storing the current volume or storing configuration changes). 
On newer installations /data already uses F2FS which is more robust for Flash memory than ext4

utmp and wtmp have very few writes, I wouldn't care for this at all. However, in general, I would not recommend SD cards for media storage at all. Put these onto a NAS, that will definitely be more robust.

soundart, 2020-07-21:
Ah ok. 

For my old use-case/system I did not notice any drawbacks having the db files not persisted on sdcard, I used /tmp for them. On reboot it scanned the library in a few seconds. Software-Volume was fixed to 100% too. The WLAN was not stable enough for NAS access, so local music library. The DAC+ PRO is always-on and gets rebooted if someone needs the plug for other purposes (vacuum cleaner, phone charger,....)  By not writing anything to flash this was robust for some years. 

But this was a special use-case with mpd as only player and with local flac files. hifiberry-os is much more feature-rich and supports many more usages! NAS is on my list too... :)

hifiberry, 2020-07-21:
For a small library this might work, an initial large library scan from a NAS can take easily longer than one hour.

soundart, 2020-07-21:
Oh wow! Yes an hour is un-acceptable boot-up time. I think I have converted my approx 200 CDs to flac using around 115GB on a "128" GB SD-Card. Network access seems to be slow or large libraries are really extremly large!!

soundart, 2020-07-22:
Hi,

I see this here on my hifiberry:
# cat /usr/lib/tmpfiles.d/samba.conf
d /var/run/samba  755 root root
d /var/log/samba  755 root root

Which sounds as it would take care of the  /var/log/samba directory. But somehow this does not seem to be active here?

hifiberry, 2020-07-22:
There is no samba running.

soundart, 2020-07-22:
Ah sorry maybe I installed non-standard packages. I tried mounting my NAS and was succesful, but I installed a samba related package for this. Now samba is not running indeed, but it seems as if it tried to do so:

```
# cat /var/log/samba/log.smbd 
[2020/07/21 15:23:49.047965,  0] ../../source3/smbd/server.c:1775(main)
  smbd version 4.11.8 started.
  Copyright Andrew Tridgell and the Samba Team 1992-2019
[2020/07/21 15:23:49.056614,  0] ../../source3/smbd/server.c:1790(main)
  error opening config file '/etc/samba/smb.conf'

```

I think I have to wipe out my sd-card (for persistent config files) and build directories and recreate my images from zero. I used "update-pi" previously.

hifiberry, 2020-07-22:
In our configuration, samba is installed, but not running. However, if you changed some stuff, it's often best to do a clean first as Buildroot doesn't detect all changes and packages that have been installed on a previous build will still be there.

soundart, 2020-07-24:
Sorry, I am currently  stuck investigating this and lost the ability to produce working images. And I did not keep my old build directories :(

What I didi I reflashed the complete sd-card with dd and recreated the build system directories on the host from scratch. 

Somehow I think that I lost an important build configuration or on the-board-runtime-configuration and I am unable to recreate it or diagnose it properly.

I noticed, that I switched from builroot-dev to buildroot-prod because ./get-buildroot decided that for me. Not sure if this is related. And I do not remember how I ended up using buildroot-dev either.

The symptons of my problem  are, that the main web interface is not running. 

In the systemd journal I see now regularly:

```
Jul 24 09:46:04 hifiberry beocreate2[13982]: /opt/beocreate/beo-extensio ns/spotifyd/index.js:56
Jul 24 09:46:04 hifiberry beocreate2[13982]:                         if (configuration.global.username && !configuration.global.username.comment && configuration.global.password && !configuration.global.password
.comment) {
Jul 24 09:46:04 hifiberry beocreate2[13982]:                                                  ^
Jul 24 09:46:04 hifiberry beocreate2[13982]: TypeError: Cannot read property 'username' of undefined
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at EventEmitter.<anonymous> (/opt/beocreate/beo-extensions/spotifyd/index.js:56:29)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at EventEmitter.emit (/usr/lib/node_modules/eventemitter3/index.js:202:33)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Object.<anonymous> (/opt/beocreate/beo-system/beo-server.js:462:8)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Module._compile (internal/modules/cjs/loader.js:1158:30)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Module.load (internal/modules/cjs/loader.js:1002:32)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Function.Module._load (internal/modules/cjs/loader.js:901:14)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)
Jul 24 09:46:04 hifiberry beocreate2[13982]:     at internal/main/run_main_module.js:18:47

```
Or maybe this isnt the main web interface and I just have to enable another package in the ./config step? Is this just me doing things in the wrong way?

hifiberry, 2020-07-24:
That's a bug in the web UI. It's fixed in our latest development tree. Alternatively, you can simply remove /etc/spotifyd.conf and update it with the default version

soundart, 2020-07-24:
Öh how do I do that?

I did:

deleted /etc/spotifyd.conf

$ git pull
Already up to date.
$ git branch
* development
$ update-pi 3 192.168.168.228

and have the same problem...

hifiberry, 2020-07-24:
Make sure you rebuild the beocreate software

./compile 3 beocreate

soundart, 2020-07-24:
Still the same:

deleted /etc/spotifyd.conf

 ./compile 3 beocreate
Building in /home/frank/projects/hifiberry-os-and-buildroot/buildroot-3
Removing /home/frank/projects/hifiberry-os-and-buildroot/buildroot-3/build/beocreate-bf70030f9360ff61a37f7669c11ddec01d143564

$ ./update-pi 3 192.168.168.228

I will do a full rebuild using ./clean now

hifiberry, 2020-07-24:
Wait. Looks like I haven't pushed the latest changes. That's not the latest beocreate version

hifiberry, 2020-07-24:
Ok, I've pushed the latest changes. Try building beocreate again

soundart, 2020-07-24:
Ok thanks

soundart, 2020-07-24:
The main web interface is online again -   but mpd is not starting. 

```
# systemctl status mpd | cat                                 
● mpd.service - mpd
     Loaded: loaded (/usr/lib/systemd/system/mpd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
TriggeredBy: ● mpd.socket

Jul 24 15:01:00 hifiberry systemd[1]: Dependency failed for mpd.
Jul 24 15:01:00 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Jul 24 15:01:00 hifiberry systemd[1]: Dependency failed for mpd.
Jul 24 15:01:00 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Jul 24 15:01:00 hifiberry systemd[1]: Dependency failed for mpd.
Jul 24 15:01:00 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Jul 24 15:01:00 hifiberry systemd[1]: Dependency failed for mpd.
Jul 24 15:01:00 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Jul 24 15:01:00 hifiberry systemd[1]: Dependency failed for mpd.
Jul 24 15:01:00 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.

```
I have these here:
```
# systemctl --failed | cat
  UNIT                                 LOAD   ACTIVE SUB    DESCRIPTION                      
● mount-data.service                   loaded failed failed Mount /data                      
● systemd-networkd-wait-online.service loaded failed failed Wait for Network to be Configured
● mpd.socket                           loaded failed failed mpd.socket                       

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

3 loaded units listed.

```

```
# systemctl status mount-data.service | cat
● mount-data.service - Mount /data
     Loaded: loaded (/etc/systemd/system/mount-data.service; static; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2020-07-24 15:01:00 CEST; 21min ago
    Process: 345 ExecStart=/opt/hifiberry/bin/mount-data.sh (code=exited, status=32)
   Main PID: 345 (code=exited, status=32)

Jul 24 15:01:00 hifiberry systemd[1]: Failed to start Mount /data.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Start request repeated too quickly.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Failed with result 'exit-code'.
Jul 24 15:01:00 hifiberry systemd[1]: Failed to start Mount /data.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Start request repeated too quickly.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Failed with result 'exit-code'.
Jul 24 15:01:00 hifiberry systemd[1]: Failed to start Mount /data.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Start request repeated too quickly.
Jul 24 15:01:00 hifiberry systemd[1]: mount-data.service: Failed with result 'exit-code'.
Jul 24 15:01:00 hifiberry systemd[1]: Failed to start Mount /data.

```
I ahve the feeling it waits for the wlan, but disabling this in the gui did not help.

/data is monuted as far as I can tell. And the network (wired) seems to be ok too.

hifiberry, 2020-07-24:
My guess is a corrupted /data. Try to run

/opt/hifiberry/bin/mount-data.sh

from command line

soundart, 2020-07-24:
This is does not give any output, but /data is there
```

# mount | grep data
/dev/mmcblk0p4 on /data type f2fs (rw,relatime,lazytime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,alloc_mode=default,fsync_mode=posix)
# systemctl status mount-data.service   ^C
# /opt/hifiberry/bin/mount-data.sh
# 

```

hifiberry, 2020-07-24:
Might be a bug of mount-data. Check the error code of mount-data

/opt/hifiberry/bin/mount-data.sh
echo $?

soundart, 2020-07-24:
# /opt/hifiberry/bin/mount-data.sh ; echo $?
0

I have this too:

```
# systemctl status systemd-networkd-wait-online.service | cat
● systemd-networkd-wait-online.service - Wait for Network to be Configured
     Loaded: loaded (/usr/lib/systemd/system/systemd-networkd-wait-online.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2020-07-24 15:26:37 CEST; 27min ago
       Docs: man:systemd-networkd-wait-online.service(8)
    Process: 143 ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --timeout=20 --any (code=exited, status=1/FAILURE)
   Main PID: 143 (code=exited, status=1/FAILURE)

Apr 17 14:37:13 hifiberry systemd[1]: Starting Wait for Network to be Configured...
Jul 24 15:26:37 hifiberry systemd-networkd-wait-online[143]: Event loop failed: Connection timed out
Jul 24 15:26:37 hifiberry systemd[1]: systemd-networkd-wait-online.service: Main process exited, code=exited, status=1/FAILURE
Jul 24 15:26:37 hifiberry systemd[1]: systemd-networkd-wait-online.service: Failed with result 'exit-code'.
Jul 24 15:26:37 hifiberry systemd[1]: Failed to start Wait for Network to be Configured.

```

and this:

# /usr/lib/systemd/systemd-networkd-wait-online --timeout=20 --any ; echo $?
0

soundart, 2020-07-24:
I rebooted (again) looks identical:
- no mpd,  
- systemctl --failed # same units as above
- zero return code for: /usr/lib/systemd/systemd-networkd-wait-online --timeout=20 --any 
-  zero return code for:/opt/hifiberry/bin/mount-data.sh ; echo $?

hifiberry, 2020-07-24:
What's the output of 

systemctl status mount-data.service

soundart, 2020-07-27:
Hi,
the output was shown here https://github.com/hifiberry/hifiberry-os/issues/123#issuecomment-663538455

I tracked this down a bit further: I checked all 4 sdcard partitions with fsck on mylaptop and the  vfat one was 'dirty' and required fixing. After repairing I finally managed to get the first sounds from the board. I then decided to restart from scratch as too many things were modified by me. 

And parts of the situation look familar. What I did:

- programmed sdcard.img with dd
- bootet hifiberry
- the initial wizard appeared, disabled wifi.
- tried to enable airplay -> beocreate2 hung
- journalctl said: Apr 17 14:37:17 hifiberry resize-partitions[251]: Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes. 
- systemctl status mount-data.service   showed this https://github.com/hifiberry/hifiberry-os/issues/123#issuecomment-663538455 again                                                                                                    
- rebootet
- beocreate2 not started, ssh into device ,something called rgnd used 400%CPU for 20seconds
- now a job "/bin/bash /opt/hifiberry/bin/detect-hifiberry " is running/hanging a subprocess " play -n synth 1 sine 1" is visible and not ending

Further infos from the cmd line:
```
# play -n synth 1 sine 1  
play FAIL sox: Sorry, there is no default audio device configured
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0 [HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
#
# wifi not disabled?
# ps -ef | grep wpa
root       190     1  0 12:50 ?        00:00:00 /usr/sbin/wpa_supplicant -u
root       222     1  0 12:50 ?        00:00:00 /sbin/wpa_supplicant -iwlan0 -c/etc/wpa_supplicant.conf
root      5776   377  0 13:22 pts/0    00:00:00 grep wpa
# systemctl status mpd | cat
● mpd.service - mpd
     Loaded: loaded (/usr/lib/systemd/system/mpd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
TriggeredBy: ● mpd.socket
# systemctl status beocreate2 | cat
● beocreate2.service - Beocreate 2 Server
     Loaded: loaded (/usr/lib/systemd/system/beocreate2.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/beocreate2.service.d
             └─override.conf
     Active: inactive (dead)
       Docs: https://github.com/bang-olufsen/create/

```

open-homeautomation, 2020-07-27:
If you're using a Pi4, this might be casue by pigpiod. Try stopping/disabling this. I'm just doing the final tests for a new pigpiod configuration on the Pi4 as the default config created lots of problems including hanging ALSA playback.

soundart, 2020-07-27:
Hi,

 it is a PI3+:
OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3

but disabling pigpiod and rebooting made beocreate2 and mpd services start...

hifiberry, 2020-07-27:
There was a lot of fiddling with the DMA settings of pigpiod. It's possible that the version you used had some settings that did not work with the Pi3,

closed 2020-09-09 by hifiberry