hifiberry/hifiberry-os #126: Spotify stop playing after random time

MBOIX, 2020-07-28:
**Describe the bug**
While playing music from Spotify Application (from my mac (electron app) or my android phone (native app client)) the music stop automatically by it self after random time.
And some times, when i want to resume playing, my hify berry device does not appear in Spotify App devices list.
It necessary to waiting some minutes in order to resume playing.

**HiFiBerryOS version**
2020052

**HiFiBerry sound card**
DAC+Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Juste playing music from OSX/Android Client.

**Expected behavior**
Never stop playing randomly.

hifiberry, 2020-07-30:
Looks like your using an older version. I recommend upgrading to the latest 20200727 version (you need to select "quick" in automatic updates).
Also make sure, you did NOT configure a username/password on Spotify. If you did, remove this.

MBOIX, 2020-08-06:
I upgrade my version with the last : 20200805.
What do you mean by i did not configure a username/password on Spotify ? 
You mean i did not configure it on the HifyBerry OS ? This is not the case for me.

Anyway, I've still the same problem, i think the issue come from my provider's router :(

MattHardwick34, 2020-08-20:
Got this issue also - randomly stops after a couple of songs and appears offline in spotify devices menu for a short while. Never had a chance to login/enter username password and am running latest OS update.

lucasso72, 2020-09-10:
Do you guys have a normal Spotify account or a family one?

MattHardwick34, 2020-09-10:
Family - but not sure if that would make a difference?

lucasso72, 2020-09-11:
I'm trying to see if there's a pattern... I also have a Family account and I've the same issue.

Bertware, 2020-09-19:
I'm having the same issue with a fresh install, Family premium account, drops out after 1 or 2 minutes. One weird thing I noticed is that the spotify app on my phone believes that I "switched" to another player (shows "playing on [other device name]"), even though it is not playing on said other device, and nobody touched anything. 

Previously I used raspotify (librespot based) on the same hardware in the same network without any issues whatsoever, so this seems like it could be spotifyd related.

Bertware, 2020-09-19:
This is a journalctl log. It seems like spotify service crashed and the system rebooted?

`Sep 19 20:18:46 living-audio-receiver systemd[1]: spotify.service: Main process exited, code=exited, status=143/n/a`

```
Sep 19 20:17:51 living-audio-receiver spotify-start[7714]: Loading <Lay Me Down> with Spotify URI <spotify:track:64GRDrL1efgXclrhVCeuA0>
Sep 19 20:17:51 living-audio-receiver spotify-start[7714]: <Lay Me Down> loaded
Sep 19 20:17:53 living-audio-receiver mpd-update-notifier[9562]: MPD error: Connection refused
Sep 19 20:18:03 living-audio-receiver mpd-update-notifier[9564]: MPD error: Connection refused
Sep 19 20:18:13 living-audio-receiver mpd-update-notifier[9567]: MPD error: Connection refused
Sep 19 20:18:23 living-audio-receiver mpd-update-notifier[9569]: MPD error: Connection refused
Sep 19 20:18:29 living-audio-receiver spotify-start[7714]: Loading <Good Job> with Spotify URI <spotify:track:3H68NURFkJOaG0woWCK2sL>
Sep 19 20:18:29 living-audio-receiver spotify-start[7714]: <Good Job> loaded
Sep 19 20:18:33 living-audio-receiver mpd-update-notifier[9574]: MPD error: Connection refused
Sep 19 20:18:34 living-audio-receiver python3[2321]: INFO: root - playback status changed from stopped to playing
Sep 19 20:18:36 living-audio-receiver python3[9611]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
Sep 19 20:18:36 living-audio-receiver python3[9611]:                                  Dload  Upload   Total   Spent    Left  Speed
Sep 19 20:18:36 living-audio-receiver python3[9611]: [158B blob data]
Sep 19 20:18:43 living-audio-receiver mpd-update-notifier[9658]: MPD error: Connection refused
Sep 19 20:18:46 living-audio-receiver /bin/cat[9680]: spotifyd still using the sound card killing it
Sep 19 20:18:46 living-audio-receiver spotify-start[7712]: Terminated
Sep 19 20:18:46 living-audio-receiver systemd[1]: spotify.service: Main process exited, code=exited, status=143/n/a
Sep 19 20:18:46 living-audio-receiver systemd[1]: spotify.service: Failed with result 'exit-code'.
Sep 19 20:18:47 living-audio-receiver python3[9683]: pgrep: no matching criteria specified
Sep 19 20:18:47 living-audio-receiver python3[9683]: Try `pgrep --help' for more information.
Sep 19 20:18:47 living-audio-receiver /bin/cat[9687]: 20200919-201834 20200919-201847 pause-all alsaloop
Sep 19 20:18:48 living-audio-receiver python3[2321]: INFO: root - playback status changed from stopped to playing
Sep 19 20:18:48 living-audio-receiver /bin/cat[9700]: 20200919-201848 20200919-201848 pause-all alsaloop
Sep 19 20:18:49 living-audio-receiver python3[2321]: INFO: root - playback status changed from playing to stopped
Sep 19 20:18:49 living-audio-receiver systemd[1]: Starting Store volume...
Sep 19 20:18:49 living-audio-receiver systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
Sep 19 20:18:49 living-audio-receiver systemd[1]: store-volume.service: Succeeded.
Sep 19 20:18:49 living-audio-receiver systemd[1]: Finished Store volume.
Sep 19 20:18:51 living-audio-receiver systemd[1]: spotify.service: Scheduled restart job, restart counter is at 3.
Sep 19 20:18:51 living-audio-receiver systemd[1]: Stopped Spotify connect.
Sep 19 20:18:51 living-audio-receiver systemd[1]: Starting Spotify connect...
Sep 19 20:18:51 living-audio-receiver systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
Sep 19 20:18:51 living-audio-receiver spotify-start[9721]: /usr/bin/spotifyd: /lib/libasound.so.2: no version information available (required by /usr/bin/spotifyd)
Sep 19 20:18:51 living-audio-receiver spotify-start[9721]: /usr/bin/spotifyd: /lib/libasound.so.2: no version information available (required by /usr/bin/spotifyd)
Sep 19 20:18:51 living-audio-receiver spotify-start[9721]: Loading config from "/etc/spotifyd.conf"
Sep 19 20:18:51 living-audio-receiver spotify-start[9721]: No proxy specified
Sep 19 20:18:51 living-audio-receiver spotify-start[9721]: Using alsa volume controller.
Sep 19 20:18:52 living-audio-receiver restore-volume[9727]: Found hardware: "HifiberryDacpAd" "" "" "" ""
Sep 19 20:18:52 living-audio-receiver restore-volume[9727]: Hardware is initialized using a generic method
Sep 19 20:18:52 living-audio-receiver restore-volume[9727]: alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock PLL:0' : Device or resource busy
Sep 19 20:18:52 living-audio-receiver restore-volume[9727]: alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DSP:0' : Device or resource busy
Sep 19 20:18:52 living-audio-receiver restore-volume[9727]: alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DAC:0' : Device or resource busy
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]: Simple mixer control 'Softvol',0
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Capabilities: volume
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Playback channels: Front Left - Front Right
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Capture channels: Front Left - Front Right
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Limits: 0 - 255
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Front Left: 255 [100%]
Sep 19 20:18:52 living-audio-receiver restore-volume[9731]:   Front Right: 255 [100%]
Sep 19 20:18:52 living-audio-receiver restore-volume[9736]: numid=21,iface=MIXER,name='ADC Capture Volume'
[... cut ...]
Sep 19 20:18:52 living-audio-receiver restore-volume[9743]:   ; Item #15 'VINR4[SE] + VINR3[SE] + VINR2[SE] + VINR1[SE]'
Sep 19 20:18:52 living-audio-receiver restore-volume[9743]:   ; Item #16 '{VIN2P, VIN2M}[DIFF]'
Sep 19 20:18:52 living-audio-receiver restore-volume[9743]:   ; Item #17 '{VIN3P, VIN3M}[DIFF]'
Sep 19 20:18:52 living-audio-receiver restore-volume[9743]:   ; Item #18 '{VIN2P, VIN2M}[DIFF] + {VIN3P, VIN3M}[DIFF]'
Sep 19 20:18:52 living-audio-receiver restore-volume[9743]:   : values=1
Sep 19 20:18:52 living-audio-receiver systemd[1]: Started Spotify connect.
```

dragotin, 2020-10-28:
I am also seeing the effect of spotify stops playing after a few minutes. 

My suspicion is that my NAS is not waking up from hibernation fast enough (it is indeed slow..) which makes mpd that is configured to use it bail out. 

This logfile snippet maybe supports that?

```
Oct 28 08:52:28 cambridge python3[332]: ERROR:dbus.proxies:Introspect error on :1.24:/org/mpris/MediaPlayer2: dbus.exceptions.DBusExceptio
n: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the 
message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Oct 28 08:52:52 cambridge beocreate2[617]: Checking MPD database update status...
Oct 28 08:52:53 cambridge python3[332]: WARNING:root:no mpris data received DBusException
Oct 28 08:53:23 cambridge python3[332]: ERROR:dbus.proxies:Introspect error on :1.24:/org/mpris/MediaPlayer2: dbus.exceptions.DBusExceptio
n: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the 
message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Oct 28 08:53:48 cambridge python3[332]: WARNING:root:got exception org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible c
auses include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, 
or the network connection was broken.
Oct 28 08:53:48 cambridge python3[332]: WARNING:root:spotifyd failed, trying to restart
```

hifiberry, 2020-10-28:
This has nothing to do with MPD. It's the Spotify player that's not running reliable. We're in the process of replacing this by another backend.

kai-roger, 2020-10-30:
Hi. I have altso seen this problem om my  newly OS updated Beocreate-amps just like in the comments above, but the most interesting is on my Amp2. When Spotify stops playing there, i have noticed that this is when my PC is sending Facebook messenger beep sounds every time i get a new message, this becouse the PC connect's to the Amp2/pi bluetooth and changes the input source. The amp2 problem i solved by deactivating bluetooth on the PC when i don't need it. As for now i'm not going to upgrade the Amp2 to 20201022 version. Awaiting future release of the OS :-)

hifiberry, 2020-11-02:
You might try the 20201101 release that comes with a new Spotify backend that seems to run more stable.

kai-roger, 2020-11-02:
So far so good :) I now see that the play/pause button in the GUI is working as expected. Before i had to press 3-4 times for it to respond.

kristojorg, 2021-03-25:
I am also seeing spotify randomly stop playing, and I have to go back into my phone and reconnect. This is what appears to be the relevant logs in my case (in reverse chronological order):
```
Mar 25 14:35:38 hifiberry vollibrespot[13634]: [Vollibrespot] : Using Alsa backend with device: default
Mar 25 14:35:38 hifiberry vollibrespot[13634]: Reading Config from "/etc/vollibrespot.conf"
Mar 25 14:35:38 hifiberry vollibrespot[13634]: vollibrespot v0.2.2 UNKNOWN UNKNOWN (librespot 1a224a3 2020-10-25) -- Built On 2021-02-03
Mar 25 14:35:38 hifiberry systemd[1]: Starting Vollibrespot...
Mar 25 14:35:38 hifiberry systemd[1]: Stopped Vollibrespot.
Mar 25 14:35:38 hifiberry systemd[1]: spotify.service: Scheduled restart job, restart counter is at 11.
Mar 25 14:35:36 hifiberry beocreate2[660]: Checking MPD database update status...
Mar 25 14:35:33 hifiberry systemd[1]: spotify.service: Succeeded.
Mar 25 14:35:33 hifiberry vollibrespot[13020]: [Vollibrespot] : SessionError: Connection reset by peer (os error 104)
Mar 25 14:35:33 hifiberry vollibrespot[13020]: [Vollibrespot] : Unable to load audio item.
```

It looks like the spotify service is restarting on a schedule. Does anyone know why that might be? Or is the SessionError what caused it to restart?

closed 2024-01-12 by hifiberry