hifiberry/hifiberry-os #127: Firmware bug and external display

lucasso72, 2020-07-30:
Hi there, 

found two issues while moving from 20200708 to 20200727. Now I see this message at boot:

[image]

...also enabling the external display doesn't work, it get stuck on "Turning External Display On" forever, a refresh on the page bring back to the main UI tho.

hifiberry, 2020-07-30:
I understand that the WebUI itself starts correctly - right?

lucasso72, 2020-07-30:
That's correct.

# cat /etc/hifiberry.state
CURRENT_PIMODEL=3B
CURRENT_MIXER_CONTROL="Digital"
CURRENT_CARD="DAC+ Pro"
CURRENT_HW="card 0"
CURRENT_HW_SHORT=0
CURRENT_SYSNAME="HiFiBerry"
CURRENT_PLAYERS=""
CURRENT_HATINFO=""
CURRENT_FEATURES="pi pi3orlater arm7 localui"
PIVERSION=3

hifiberry, 2020-07-30:
Ok,m try enabling the local UI using

systemctl enable --now cog.service weston.service

lucasso72, 2020-07-30:
the command get stuck there...

hifiberry, 2020-07-30:
Check the logs:

systemctl enable --now cog.service weston.service & journalctl -f

lucasso72, 2020-07-30:
# systemctl enable --now cog.service weston.service & journalctl -f
[1] 1014
-- Logs begin at Fri 2020-04-17 14:37:13 CEST. --
Jul 30 12:19:30 hifiberry systemd[1]: Condition check resulted in Platform Persi                  stent Storage Archival being skipped.
Jul 30 12:19:30 hifiberry systemd[1]: Condition check resulted in Update is Comp                  leted being skipped.
Jul 30 12:19:59 hifiberry systemd[1]: Reloading.
Jul 30 12:20:00 hifiberry systemd[1]: /usr/lib/systemd/system/dbus.socket:5: Lis                  tenStream= references a path below legacy directory /var/run/, updating /var/run                  /dbus/system_bus_socket → /run/dbus/system_bus_socket; please update the unit fi                  le accordingly.
Jul 30 12:20:00 hifiberry systemd[1]: /usr/lib/systemd/system/winbind.service:9:                   PIDFile= references a path below legacy directory /var/run/, updating /var/run/                  samba/winbindd.pid → /run/samba/winbindd.pid; please update the unit file accord                  ingly.
Jul 30 12:20:00 hifiberry systemd[1]: /usr/lib/systemd/system/smb.service:10: PI                  DFile= references a path below legacy directory /var/run/, updating /var/run/sam                  ba/smbd.pid → /run/samba/smbd.pid; please update the unit file accordingly.
Jul 30 12:20:00 hifiberry systemd[1]: /usr/lib/systemd/system/samba.service:10:                   PIDFile= references a path below legacy directory /var/run/, updating /var/run/s                  amba/samba.pid → /run/samba/samba.pid; please update the unit file accordingly.
Jul 30 12:20:00 hifiberry systemd[1]: /usr/lib/systemd/system/nmb.service:10: PI                  DFile= references a path below legacy directory /var/run/, updating /var/run/sam                  ba/nmbd.pid → /run/samba/nmbd.pid; please update the unit file accordingly.
Jul 30 12:20:01 hifiberry systemd[1]: Condition check resulted in Platform Persi                  stent Storage Archival being skipped.
Jul 30 12:20:06 hifiberry beocreate2[524]: Checking MPD database update status..                  .
Jul 30 12:20:09 hifiberry systemd[1]: Reloading.
Jul 30 12:20:10 hifiberry systemd[1]: /usr/lib/systemd/system/dbus.socket:5: Lis                  tenStream= references a path below legacy directory /var/run/, updating /var/run                  /dbus/system_bus_socket → /run/dbus/system_bus_socket; please update the unit fi                  le accordingly.
Jul 30 12:20:10 hifiberry systemd[1]: /usr/lib/systemd/system/winbind.service:9:                   PIDFile= references a path below legacy directory /var/run/, updating /var/run/                  samba/winbindd.pid → /run/samba/winbindd.pid; please update the unit file accord                  ingly.
Jul 30 12:20:10 hifiberry systemd[1]: /usr/lib/systemd/system/smb.service:10: PI                  DFile= references a path below legacy directory /var/run/, updating /var/run/sam                  ba/smbd.pid → /run/samba/smbd.pid; please update the unit file accordingly.
Jul 30 12:20:10 hifiberry systemd[1]: /usr/lib/systemd/system/samba.service:10:                   PIDFile= references a path below legacy directory /var/run/, updating /var/run/s                  amba/samba.pid → /run/samba/samba.pid; please update the unit file accordingly.
Jul 30 12:20:10 hifiberry systemd[1]: /usr/lib/systemd/system/nmb.service:10: PI                  DFile= references a path below legacy directory /var/run/, updating /var/run/sam                  ba/nmbd.pid → /run/samba/nmbd.pid; please update the unit file accordingly.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Platform Persi                  stent Storage Archival being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Huge Pages Fil                  e System being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Rebuild Journa                  l Catalog being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Commit a trans                  ient machine-id on disk being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Load Kernel Mo                  dules being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in FUSE Control F                  ile System being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Rebuild Hardwa                  re Database being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Platform Persi                  stent Storage Archival being skipped.
Jul 30 12:20:11 hifiberry systemd[1]: Condition check resulted in Update is Comp                  leted being skipped.

Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Journal Audit Socket being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Huge Pages File System being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Rebuild Journal Catalog being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Load Kernel Modules being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in FUSE Control File System being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Condition check resulted in Update is Completed being skipped.
Jul 30 12:20:31 hifiberry systemd[1]: Starting Store volume...
Jul 30 12:20:31 hifiberry systemd[1]: store-volume.service: Succeeded.
Jul 30 12:20:31 hifiberry systemd[1]: Finished Store volume.

lucasso72, 2020-07-30:
[hifiberry-debug.zip](https://github.com/hifiberry/hifiberry-os/files/5000626/hifiberry-debug.zip)

hifiberry, 2020-07-30:
Might be a bug. Will look into this, but it's gonna take some time. I recommend to go back to the previous version if you need the local UI.

lucasso72, 2020-07-30:
No big deal, thanks for fast response, as usual. 
Btw I noticed this release being MUCH faster to boot, either you cleared some bugs or .... it miss to load something :)

hifiberry, 2020-07-30:
Looks like the same changes that improved startup time messed up some other stuff ;-) We did some major changes in the boot process. But it looks like it's not too hard to fix.

tompeel, 2020-07-30:
I promise to not start messaging on both platforms 😂
In order to speed up boot, i wondered if you can boot HifiberryOS from USB on the 3 and 4?

hifiberry, 2020-07-30:
No, USB boot isn't supported. It's rather new on the Pi4 and it will require quite a lot of changes on different places to get this working.

hifiberry, 2020-07-31:
If somebody wants to try an experimental build, you can try uprading from command line

/opt/hifiberry/bin/update --experimental

Note that I couldn't test this build, there might still be issues...

lucasso72, 2020-07-31:
So... with the experimental build the external display is back to life, still have that message I screenshot at the beginning here "firmware bug".

lucasso72, 2020-07-31:
Long living bugs:
Shutdown most of the times actually reboot.
Album covers often doesn't show on the external screen but does on the webUI, most of the times with Spotify.

hifiberry, 2020-08-04:
You can simply ignore the "firmware bug" message. We'll package a new release soon with the display fix.

hifiberry, 2020-08-05:
The bugfix release is out now:
https://github.com/hifiberry/hifiberry-os/releases/tag/v20200805

closed 2020-08-05 by hifiberry