hifiberry/hifiberry-os #128: HiFiBerryOS fresh install won't complete boot

haylingsailor, 2020-08-03:
**Describe the bug**
On a fresh install of HiFiBerryOS for Raspberry Pi Version 4, first boot stuck on endless loop (waited an hour...) Not fixed by off/on again...

The following sequence repeats:
- Starting Store volume
Transient message:
- 'A start job is running for Starts Psplash Boot screen'
- Finished Store volume.

**HiFiBerryOS version**
20200727

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. New Raspberry Pi 4B 2GB
2. Add Amp2
3. Follow [SD Card instructions](https://www.hifiberry.com/docs/hifiberryos/setup-hifiberryos/) - Use Etcher & new SD card
4. Install SD card 
5. Connect to LAN. 
6. Connect to monitor
7. Power on.

**Expected behavior**
HiFiBerryOS starts up.

**Screenshots**
https://photos.app.goo.gl/S5Ev981D6EAMUsWJ9

**Additional context**
This is a new purchase of an AMP2 bundle - no previous use of any components (sd card, Pi)

haylingsailor, 2020-08-03:
If I install Raspian instead (using the [Raspberry Pi Imager for windows](https://www.raspberrypi.org/downloads/), then the system boots up fine into a desktop. So the Raspberry Pi & SD card are ok.

haylingsailor, 2020-08-03:
The [20200708](https://github.com/hifiberry/hifiberry-os/releases/tag/v20200708) version worked. So I'll use that for now.

hifiberry, 2020-08-04:
We'll publish a new release soon that will fix some bugs. Even tough, we haven't seen this behavior with the latest release here.

hifiberry, 2020-08-05:
While we can't reproduce this behavior in the lab, you might check the latest release:
https://github.com/hifiberry/hifiberry-os/releases/tag/v20200805

closed 2020-09-09 by hifiberry