hifiberry/hifiberry-os #138: Does HiFiBerry OS have all the latest RPi4 firmware updates - Heat?

ctbarker32, 2020-09-02:
Do the latest builds of HiFiBerry OS have the latest RPi 4 firmware updates? Specifically, to address the excessive Heat of the RPi4?

I recently bought the new DAC2 and the white steel case and was alarmed at how hot this combo is. HFB tech support say everything is just fine but I am unsure?

https://maker.pro/raspberry-pi/tutorial/raspberry-pi-4-firmware-updates-help-prevent-overheating

hifiberry, 2020-09-02:
The link you've posted is about Raspbian. We're not using Raspbian. The Pi4 runs hotter than the Pi3, that's normal.

closed 2020-09-02 by hifiberry

ctbarker32, 2020-09-02:
Raspian or not. You're using the linux kernel and should have the firmware updates?

hifiberry, 2020-09-02:
Kernel is 4.9.x.
We don't use "latest" kernels, but kernels that have been tested to be stable.

ctbarker32, 2020-09-02:
I am not asking for "beta kernels. I just want to know if all approved/released updates are applied to take advantage of any heat management solutions. My HFB DAC2 is too hot to touch around the USB ports. You seem annoyed that I am asking a reasonable but important question?

ctbarker32, 2020-09-02:
I think you mean Kernel 4.19?

hifiberry, 2020-09-02:
Yes, I mean 4.19. 

There are no "approved" updates. 

And no, we're not using the latest "released" kernel (that changes almost every week), but one that has been tested by us to work. There are major changes from 4.19 to 5.4 and we'll integrate this once everything is ready and working fine.

You can find the firmware version that's used in buildroot:
https://github.com/buildroot/buildroot/blob/2020.05.x/package/rpi-firmware/rpi-firmware.mk

ctbarker32, 2020-09-02:
Thank you for the information. I really do like what are doing with HiFiBerry OS and have been suggesting it to fellow Audiophiles.

https://www.youtube.com/watch?v=2k0o0Uy9CKQ&t=1950s

hifiberry, 2020-09-02:
Thanks for this!