hifiberry/hifiberry-os #141: External display not working

lucasso72, 2020-09-09:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
After update to latest version  the touchscreen of the external display is not working anymore, it's stuck in the "now playing" screen tho it's being updated with current song.

**HiFiBerryOS version**
20200908

**HiFiBerry sound card**
CURRENT_PIMODEL=3B
CURRENT_MIXER_CONTROL="Digital"
CURRENT_CARD="DAC+ Pro"
CURRENT_HW="card 0"
CURRENT_HW_SHORT=0
CURRENT_SYSNAME="HiFiBerry"
CURRENT_PLAYERS=""
CURRENT_HATINFO=""
CURRENT_FEATURES="pi pi3orlater arm7 localui"
PIVERSION=3

**To Reproduce**
See "Describe the bug" section.

**Expected behavior**
See "Describe the bug" section.

**Screenshots**
See "Describe the bug" section.

**Browser (if applicable)**
N/A

**Additional context**
Add any other context about the problem here.

lucasso72, 2020-09-14:
anyone to acknowledge this?

hifiberry, 2020-09-14:
Can't see this here. However, I don't have a touchscreen to test with.

lucasso72, 2020-09-14:
anything I can provide to help troubleshoot?

lucasso72, 2020-09-14:
On Raspberrypi.org I found this command the check the touchscreen has been detected:

` dmesg | grep -i ft5406`

and in my case it doesn't return what's expected. I'm trying to understand if it's an hw issue but it's strange because it was working until the moment I upgraded to latest hifiberry release. I just tried to go back to previous version and the problem remain...

lucasso72, 2020-09-16:
Guys I checked by installing Raspberry pi OS and the touch screen is working, so it's not an hw issue. Also is strange that reverting the Hifiberry back to the previous installed version, keep having the issue.

hifiberry, 2020-09-17:
Might be a problem with the firmware on the latest version:
https://support.hifiberry.com/hc/en-us/community/posts/360012803457

lucasso72, 2020-10-12:
Any plan to release a build that fixes this issue? tks.

hifiberry, 2020-10-15:
This will be fixed in the next release

hifiberry, 2020-10-22:
@lucasso72 I think this is fixed now - correct?

lucasso72, 2020-10-23:
Correct! thanks.

closed 2020-10-23 by hifiberry