hifiberry/hifiberry-os #142: dev branch build failure, nodejs-12.18.0

soundart, 2020-09-16:
**Describe the bug**

build stops with error message:
`
+ pigpio@3.2.3
added 371 packages from 251 contributors in 43.566s
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-x64/electron-napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-x64/node-napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/bufferutil/prebuilds/linux-x64/electron-napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/bufferutil/prebuilds/linux-x64/node-napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
make[1]: *** [package/pkg-generic.mk:362: /home/frank/projects/hifiberry-os-and-buildroot/buildroot-3/build/nodejs-12.18.0/.stamp_installed] Error 1
make: *** [Makefile:84: _all] Error 2
`

**HiFiBerryOS version**

hifiberry-os: fa2c9a319b16b4987653bf23e3f86e846b40ae63

buildroot-2020.08

**HiFiBerry sound card**

DAC+ PRO

**To Reproduce**
I followed the steps in doc/setupdev.md

**Expected behavior**

exit 0

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**

The files listed in the error message  are not from my system probably:

$ ls /usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-x64/electron-napi.node
ls: cannot access '/usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-x64/electron-napi.node': No such file or directory

hifiberry, 2020-09-16:
Yes, that's a known problem. Just start compile again and it will go on.

hifiberry, 2020-09-16:
BTW: Current development tree has been changed for buildroot 2020-08 already. You might have to get the latest version using get-buildroot-prod

soundart, 2020-09-16:
Ah thanks. I switched the buildroot today to 2020-08.

hifiberry, 2020-09-16:
Make sure, you run the script as it patches a few things...

soundart, 2020-09-16:
The build has finished after restarting now, thank you!

closed 2020-09-16 by soundart