hifiberry/hifiberry-os #143: Popping noise when skipping Spotify songs

Bertware, 2020-09-19:
**Describe the bug**
When playing spotify and jumping to the next track (either because the previous track ended or the user pressed the skip button in the Spotify app), the sound card is closed and opened again. This causes a popping noise on the speakers.

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

**HiFiBerry sound card**
DAC+ADC Pro combined with the AAMP60

**To Reproduce**
Steps to reproduce the behavior:
1. Play a spotify song
2. Skip to the next song
3. You can see the output led on the DAC+ADC going out before it goes on again after a few hundred milliseconds. A popping sound can be heard in the speakers.

**Expected behavior**
The output stream remains open in order to prevent the DAC+ADC from muting/shutting down inbetween songs

**Additional context**
This is likely related to the spotifyd client closing the output stream. Similar popping occurs as well with the analog input (when there is no input for a second or maybe even sooner). One solution would be fixing this in the spotify client configuration/music source, but this might be solved by adding a "time before mute" option or something similar which would keep the output device active for a set number of seconds even if no music is playing.

hifiberry, 2020-09-21:
We probably can't do anything about this as this is the behaviour of Spotifyd.

dwsch, 2021-05-04:
I have the same effect/behavior, but when I use Roon (I don't use Spotify). When I skip to another song or the next song is played automatically, a pretty loud popping sound can be heard.

- HiFiBerry HD2 with RasPi4 and RasPi Touchscreen
- OS 20210408

hifiberry, 2021-05-04:
I don't see this here. What amplifier are you using?

hifiberry, 2021-05-07:
As I don't see this here, there is probably nothing we can do about it.

3579km, 2024-02-26:
Observing during Prev/Next and Pause tracks. Need something like fade-out effect.
DAC2 HD with DSP board on Raspberry Pi 4, WiFi connection

arnold-graf, 2024-08-18:
I’m observing the same behaviour with the DAC2 Pro, Apple Music via Airplay, and an optical connection