hifiberry/hifiberry-os #144: Popping noise during short pauses when using analog in

Bertware, 2020-09-19:
**Describe the bug**
When using the analog input for a TV signal, short silences cause the output stream to shutdown within a second. This causes a pop in the speakers.

**HiFiBerryOS version**
20200908

**HiFiBerry sound card**
Hifiberry DAC+ADC pro with AAMP60

**To Reproduce**
Connect a TV to the analog input. Set the analog input to automatically start when a signal is detected. Browse calmly through a game console menu, such as the Xbox One interface or Nintendo Switch interface. Alternatively, pause a movie for a second or two. The audio stream is closed (led on the DAC+ADC pro goes out) and a pop can be heard.

**Expected behavior**
During short pauses (browsing through a menu in a game console, pausing a movie for a few seconds, ...),  the audio output stream remains open. 

**Additional context**
This could be a slider in the analog-in configuration: "Wait time before disconnect: x seconds". In this case, the user should be able to choose a duration between 1 (or even 0?) seconds and 120 seconds, as short silences are normal when gaming or starting a new Netflix episode. 

Other than this, I'd just like to note that I'm impressed with how smooth hifiberry-os works, issues like these are just the last finishing touches. Thanks for all the hard work!

Bertware, 2020-09-19:
To elaborate a bit more on this, sliders for sensitivity would be useful as well. For some reason, I experience quick successions of starts and stops (quick popping which sounds more like rattling, sounds like about 5 mutes/unmutes in +-0.5 seconds), even with the sensitivity on low. Being able to configure the sensitivity manually (input level + minimum input time) would resolve this, possible as a hidden "advanced" settings. I assume the low/medium/high sensitivity settings do the same already, only hardcoded

hifiberry, 2020-09-21:
This is not a bug, but a feature request. Adding a "minimum input time" would complicate the code quite a lot. Not sure if we're going to implement something like this in the future. 
If somebody wan'ts to look into this, the source for this player component is here:
https://github.com/hifiberry/alsaloop

hifiberry, 2020-11-07:
This has been implemented now by @Bertware and will be included in the next release

closed 2020-11-07 by hifiberry