hifiberry/hifiberry-os #146: Hifiberry only?

JasonSwindle, 2020-09-29:
Howdy,

I could not find a clear yes / no on the marketing page or docs, but is this only made to support Hifiberry products? Or will other rPi amp / dac work too?

hifiberry, 2020-09-29:
It works only with HiFiBerry products. Have a look at our portfolio and you will probably find the right HiFiBerry board for you:
https://www.hifiberry.com/shop#boards

closed 2020-09-29 by hifiberry

imgrant, 2020-09-29:
Slightly disingenuous, it works just fine with many other-brand DAC and amp HATs for the raspberry pi, because they use exactly the same TI chip(s), and so use exactly the same hifiberry-named DTB overlay(s).
That's not to say that all features work, so, e.g. the DSP functionality which requires one of the DSP-enabled boards (which, as far as I'm aware, only HiFiBerry make) might not work.
It's also worth pointing out that HiFiBerry are arguably pioneers in this space (e.g. they wrote the requisite DTB overlays, and hence why they're named so), but all that aside, technically, yes, HiFiBerryOS works with other-brand HATs.