hifiberry/hifiberry-os #147: "configure network using WiFi setup", but no SSID is broadcasted

febs, 2020-09-29:
**HiFiBerryOS version**
I can't get to access it, so I can't tell. I downloaded the image today though (29-09-2020) from the official websige.

**HiFiBerry sound card**
DAC zero

**To Reproduce**
Steps to reproduce the behavior:
Boot the system on a raspberry zero W.
After setting itself up, the system consistently shows 'no network connected, configure network using WiFi setup"

**Expected behavior**
A wifi network should be broadcasted as per documentation, but none appears.

Thanks,

hifiberry, 2020-10-15:
We don't test the Pi Zero release. Can you see the same problem with a Pi3 or Pi4?

druilaap, 2020-10-17:
I have exactly the same problem. Downloaded the newest version today and get the same error on a pi zero W. Is there a way to start a terminal using a keyboard?
Volumio did start a wifi hotspot on this pi zero W.

druilaap, 2020-10-17:
I manually configured the wpa_supplicants.conf and moved the interfaces.bak to interfaces. The pi0w now connects to the wifi, and claims to be fully booted with "Web interface ready at http://192.168.x.x" but the http server on port 80 is not up.

druilaap, 2020-11-23:
Anything I can do to help debug this?

usul27, 2020-11-24:
Problem is that you can't really debug anything on a system that has no network connectivity.

febs, 2020-11-25:
Well that does actually have WiFi and Bluetooth. It is a raspberry pi W.
Apologies if that was implicit.

st0nec0ld, 2021-01-05:
Any news?

With 20201213-pi0w.img it still dont broadcast the wifi setup. With Volumio it works.

hifiberry, 2021-01-05:
No, Pi Zero releases are untested and we don't debug stuff on the Pi Zero. We might even consider not releasing Pi Zero releases in the future. It's just too much hassle and the Pi Zero isn't the best product for this.

st0nec0ld, 2021-01-05:
Tested it again, now it works. I dont no why, works with 20201312-pi0w.img

HIFIBERRY DAC+ ZERO is an official product you can buy on https://www.hifiberry.com/shop/, dont stop releasing pi zero releases.

hifiberry, 2021-01-05:
Sure, you can buy the DAC+ Zero, but that doesn't mean, we have to support the Pi Zero in HiFiBerryOS. We simply think the limited speed and memory of the Pi Zero don't make it the right choice for HiFiBerryOS. For the same reason we don't support the old B+ models anymore.

ralphkok, 2021-02-05:
It would be good if you would mention this on your product descriptions then. I just bought a DAC Zero along with a RPi zero, just to now learn that you don't actually support it.
Could you point to solutions to piping audio from the DAC Zero inputs using regular Raspbian?

usul27, 2021-02-05:
This is clearly shown on the HiFiBerryOS web site. Feel free to use any other distribution that you like. There are no inputs on the DAC Zero, this is an output device.

0bmxa, 2021-02-06:
I was having the same issue on my Pi Zero W, so I investigated a bit.

It seems, `beocreate2.service` is failing to launch because the `/usr/bin/node` binary has an incredibly long startup time on the Pi Zero W. Already just `time node --help` reports 3.5 mins(!). Similar around 3:10 mins for a simple `console.log('hello world')`.
Launching `/usr/bin/node /opt/beocreate/beo-system/beo-server.js vv` manually also takes around 3-4mins, but seems to work quite fine after.

Therefore, I now edited the `beocreate2.service` on my system to include a `TimeoutStartSec=900` (not sure if it needs to be that high), so systemd won't kill & retry after the default timeout (of 90s or so?), and voila it's slow but it works! :D

Also, after beocreate2 finished launching, the WiFi setup AP is also available.

@hifiberry I totally understand you don't want to actively support the Pi Zero, but do you think adding a `TimeoutStartSec` to [beocreate2.service](https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/beocreate/beocreate2.service) is an option? If you want I can make a PR!

usul27, 2021-02-07:
Sure, we're happy to integrate a PR. However, globally setting the timeout for all systems is not good as it might have negative effects on other platforms as restarts will take much longer. I think it's best to add the timeout as an parameter in the package configuration. This way we can use different configurations for different PI's

0bmxa, 2021-02-07:
Thanks for the feedback @usul27! I absolutely agree, this should only be set so high for the Pi Zeros. 
I made a draft here https://github.com/hifiberry/hifiberry-os/pull/235, to be able to comment the code directly.

closed 2023-04-04 by hifiberry