hifiberry/hifiberry-os #150: No input field for login to Spotify

tatobrutto, 2020-10-13:
**Describe the bug**
When selecting the Spotify the input fields for credential are missing.

**HiFiBerryOS version**
20000908

**HiFiBerry sound card**
Beocreate

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'inputs menu'
2. Click on 'spotify'
3. See error

**Expected behavior**
I'm using one of the previous HifiBerryOs image was on my p and connceting to Spotify was working perfectly with it. Unfortunately that version had another bug with volume too low even at 100%.
I would kindly suggest to allow users to download older images if they want.

**Browser (if applicable)**
 - Linux Mint
 - Firefox (latest)

**Additional context**
Add any other context about the problem here.

hifiberry, 2020-10-13:
This has been removed as spotifyd doesn't run stable with this setting. It's not a bug, but expected to behave like this. Spotify will work without a username/password.

tatobrutto, 2020-10-13:
Where I can find information about setting up / use correctly Spotify with no username/password?

hifiberry, 2020-10-13:
There is nothing do setup. Just enable Spotify

tatobrutto, 2020-10-13:
I did it. But I was not able to use it. Even after logging in my spotify account, the only devide "hifiberry" was the old one I created from the older version of HifiberryOS. Could you explain the "magic"? How Spotify will ever know that this HifiberryOS belongs to me without enter at least an account? Or, should I input some kind of ID in the Spotify website?

hifiberry, 2020-10-13:
You simply need to use your Spotify app on your smartphone or PC on the same network. Spotify detects devices on the same network.

tatobrutto, 2020-10-13:
I will try againg later and, if the case, open another issue regarding the use of Spotify.

closed 2020-10-13 by tatobrutto

hifiberry, 2020-10-13:
If you have specific questions about a feature, do not open a bug report, but post at the HiFiBerry forum:
https://support.hifiberry.com/hc/en-us