hifiberry/hifiberry-os #156: DLNA not working on 20201020

schwachstelle, 2020-10-21:
**Describe the bug**
When DLNA is enabled through the web gui, it turns off a few seconds (~10) later.

**HiFiBerryOS version**
20201020

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. go to 'General' and turn on MPD
2. go to 'Sources' and turn on DLNA
3. Wait 10 seconds
4. go to any other page, e.g. start page ('Hifiberry')
5. go to Sources' again and see: DNA is switched off

**Expected behavior**
DLNA should stay switched on and not turn off automatically.

**Screenshots**
no screenshot

**Browser (if applicable)**
 - OS: Linux
 - Browser Firefox
 - Version 81.0

**Additional context**
running an a PI 4.

hifiberry, 2020-10-22:
I can confirm this bug. We'll fix this in the next bugfix release.

schwachstelle, 2020-12-23:
Fixed!  :-)

closed 2020-12-23 by schwachstelle