hifiberry/hifiberry-os #157: Radio cannot be stopped on 20201020

schwachstelle, 2020-10-21:
**Describe the bug**
When playing a radio station, after some time the stop button on the web gui doesn't work anymore and even other sources cannot interrupt radio playing. 

**HiFiBerryOS version**
20201020

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Play a radio station. I switched stations several times, don't know if this is relevant.
2. wait some time (maybe one hour)
3. press stop button on the web gui -> radio doesn't stop
4. try to play a song through Spotify connect on HiFiBerry (in my case from the official Spotify Android app) -> app indicates HiFiBerry is actually playing, but in fact radio keeps playing.

**Expected behavior**
Stop button should cause radio to stop. Switching to Spotify while playing radio should also stop radio and play Spotify instead.

**Screenshots**
---

**Browser (if applicable)**
 - OS: Linux
 - Browser Firefox
 - Version 81.0

**Additional context**
running on a Pi 4

hifiberry, 2020-10-22:
Unfortunately I can't reproduce this problem here. 

Can you try the following when this happens again:

- login via SSH
- run 
   /opt/hifiberry/bin/pause-all Test

Let us me know if this stops the radio playback and if there's any output on the console.

schwachstelle, 2020-10-22:
I cannot easily retry this at the moment. My Pi 4 is doing his normal work again and my Pi 2 is on 20200908 with the device tree overlay files patched (#158). I will try to reproduce the problem on the Pi 2 when the bugfix release for 20201020 is available.

schwachstelle, 2020-10-25:
I had this problem again: after playing several different radio stations for a while, I started Spotify on my Linux box, switched output device to HiFiBerry and pressed play. Though Spotify showed it was actually playing (moving progress bar) on HiFiBerry, radio was playing on.
I ran '/opt/hifiberry/bin/pause-all Test', this took ~5 seconds and than paused the radio station and Spotify played.

hifiberry, 2020-10-26:
This looks like a problem with the Spotify backend. We're working on a completely new implementation of this. Therefore, I won't look into this anymore. The new backend that should be available in the next release should work better.

hifiberry, 2020-10-27:
Can you try the experimental release 20201027?

Just log in and run
/opt/hifiberry/bin/upgrade --experimental

schwachstelle, 2020-10-27:
With 20201027 this works much better. Switching from radio to Spotify (without stopping radio first) takes 6 seconds, but it works. Switching from Spotify to radio (without stopping Spotify first) does not really work. Spotify goes on for ~ 30 seconds and then stops, but radio doesn't start. I can start radio afterwards. So It's not perfect yet, but much better than before.

But this is not the only improvement in 20201027:
 * the remote control works flawless now with Spotify
 * Pi 2 finally gets detected correctly   :-)

hifiberry, 2020-10-27:
Ok, still room for improvement ;-) 
Problem when switching sources is it to stop the previous source which can be quite difficult on some sources...

hifiberry, 2020-10-28:
Found a way to do a faster check for the active player. Will integrate this into the next release.

hifiberry, 2020-11-01:
/opt/hifiberry/bin/update --experimental

schwachstelle, 2020-11-01:
I just updated to 20201101; I suppose this is the same version as I would get with /opt/hifiberry/bin/update --experimental !?

schwachstelle, 2020-11-01:
With 20201101, switching from Spotify to radio still doesn't work as expected. When i click on a favorite radio station, the web gui indicates it would be playing the radio station, but in fact Spotify plays on.

schwachstelle, 2020-11-01:
... at least sometimes.

hifiberry, 2020-11-08:
Still found some bugs that have now been fixed. These fixes will be integrated in the next release.

hifiberry, 2020-12-14:
Some patches have been integrated in 20201202 and 20201213. I don't see this problem here. Can you try the latest 20201213 release?

hifiberry, 2021-01-20:
As there was no feedback, I'm gonna close this.

closed 2021-01-20 by hifiberry