hifiberry/hifiberry-os #158: Images for Pi 2 don't contain the HiFiBerry device drivers

schwachstelle, 2020-10-21:
**Describe the bug**
The images for the Pi 2 don't contain the HiFiBerry device drivers. The directory /boot/overlays with the device tree overlays is missing. Therefore (most) HiFiBerry cards cannot be used.

**HiFiBerryOS version**
20201020, 20200908

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. flash HiFiBerryOS image to SD card
2. mount first partition
3. cd to <mount point>/boot
4. ls -l
5. there is no directory 'overlays'

**Expected behavior**
the device tree overlays should be included in the images for the Pi 2.

**Screenshots**
no screenshot

**Browser (if applicable)**
no browser

**Additional context**
surprisingly, the images for the Pi 2 *do work* in conjunction with a HiFiBerry DAC+ PRO. I don't know why.

hifiberry, 2020-10-22:
I can confirm this bug. We'll fix this in the next bugfix release.

hifiberry, 2020-10-22:
Looks like you found a bug that's there for a very long time.
https://bugs.busybox.net/show_bug.cgi?id=13281

hifiberry, 2020-10-22:
BTW: The overlay files are not really the device drivers, but only a hardware description that's required to load the driver. The DAC+ Pro has this also in the EEPROM on the card.

schwachstelle, 2020-10-22:
Ok, looking forward for the bugfix release.

Until then, if anyone is interested, a quick fix. You will need a linux box for this. Be sure to check device names and mount points for your setup. You'll have to become root for at least the mount commands.

1. switch to /tmp:
cd /tmp
1. prepare two mountpoints:
mkdir pi2 pi4
1. take  SD card from the Pi 2 and mount it:
mount /dev/mmcblk0p1 pi2
1. get the HiFiBerryOS image zip for the Pi 4:
wget https://www.hifiberry.com/images/hifiberryos-pi4.zip
1. unzip the Pi 4 image:
unzip hifiberryos-pi4.zip
1. mount the boot partition from the Pi 4 image:
mount -o loop,ro,offset=512 hifiberryos-20201020-pi4.img pi4
1. copy the device tree overlays from the Pi 4 image to the Pi 2 SD card:
cp -R pi4/overlays pi2
1. unmount filesystems:
umount pi2 pi4
1. put the SD card into your Pi 2. HiFiBerryOS should now be able to detect the HiFiBerry board.

hifiberry, 2020-11-07:
This has been fixed in 20201101

closed 2020-11-07 by hifiberry