hifiberry/hifiberry-os #159: Wrong current playing info

armatron, 2020-10-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When stop BT streaming from phone and start playing from another source, current play info doesn't change

**HiFiBerryOS version**
20200908

**HiFiBerry sound card**
Digi+

**To Reproduce**
1. Connect phone via BT to HFiBerryOS
2. Start streaming. Source and current music information is displayed in the screen where are pause, next and previous buttons.
3. Stop streaming.
4. Disconnect phone from BT connection
5. Go to Sources and select any radio station for example
6. Source and current music information does not change. It shows last BT streamed music information.

**Expected behavior**
Should display radio station as normally when no BT device is connected after HiFiBerryOS restart.

**Additional context**
Browser refresh does not help. Only HiFBerryOS restart helps.

hifiberry, 2020-10-28:
Unfortunately I can't reproduce this specific problem here. However, there are still some problem when switching between players. Maybe future bug fixes here will also fix this.

schwachstelle, 2020-11-02:
With 20201101 sometimes the cover art changes after a few seconds. I suppose this is a result of multiple sources being queried. But I think this should not happen.

hifiberry, 2020-11-02:
It should depending on the cover art settings. Check your configuration under "General"/"Display"

armatron, 2020-11-12:
Bug, posted in original post is still there. Tested in current version 20201101

hifiberry, 2020-12-14:
Some patches have been integrated in 20201202 and 20201213. I don't see this problem here. Can you try the latest 20201213 release?

armatron, 2020-12-15:
I updated HiFiBerryOS from 20201202 to 20201213. 
After connecting my phone to HiFiBerryOS via BT, volume slider remains in the previous position, which is the same as before. This seems normal behaviour.  When start streaming from my phone, volume slider in HiFiBerryOS goes to maximum. Changing volume on my phone does not affect volume in HiFiBerryOS like before (volume is not synchronised).
After reconnecting my phone to HiFIBerryOS and start streaming from my phone, there is no audio on my speakers.
Playing info does not change after first streamed song even If I start stream another music or start playing internet radio on HiFIBerryOS. The same issue is in previous versions.

I think that version 20201202 was better because volume was synchronised and was no problems when streaming after reconnecting.

hifiberry, 2020-12-15:
Unfortunately I can't reproduce this behavior here. What OS and version are you using on your mobile phone?

armatron, 2020-12-15:
Sony Xperia XZ1, Android 9, streaming from native Music app.

hifiberry, 2021-01-20:
Duplicate of https://github.com/hifiberry/hifiberry-os/issues/148

closed 2021-01-20 by hifiberry