hifiberry/hifiberry-os #160: rebooting loop after update to 20201022

sroeglinger, 2020-10-26:
**Describe the bug**
After updating Hifiberry to 20201022 the device is in a reboot loop and cannot be used anymore.
Installing Hifiberry 20201022 on the sd-card from zip-file shows the same issue.
Installing Hifiberry 20200908 on the sd-card works perfectly.

**HiFiBerryOS version**
20201020 and 20201022

**HiFiBerry sound card**
DAC2 HD

**To Reproduce**
Steps to reproduce the behavior:
1. install 20201022 on sd-card
2. boot device 

**Expected behavior**
The WebUI cannot be accessed.
A connection via ssh is disconnected after some time.

**Screenshots**
no screenshots available

**Browser (if applicable)**
-

**Additional context**
no additional context relevant

hifiberry, 2020-10-26:
Yes, this is a known bug. We're working on a bugfix release. As a workaround, create a file noreboot on the FAT partition and revert to the latest version after the system has been booted.

sroeglinger, 2020-10-26:
Thanks for your help.
I will use 20200908 until an update is available.

hifiberry, 2020-10-26:
After downgrading to 20200908, you might try the experimental build that should fix this by running

/opt/hifiberry/bin/update --experimental

(you can even try it from the 20201020 when you can log in)

sroeglinger, 2020-10-26:
looks quite good so far 👍

hifiberry, 2020-10-26:
20200908 or the experimental?

sroeglinger, 2020-10-26:
At the moment I am listing to music with Hifiberry 20201026.
And there is not workaround applied (noreboot file on FAT partition).

hifiberry, 2020-10-26:
Thanks. If we don't find any other bug, this will probably become the "official" release.

hifiberry, 2020-11-06:
As 20201101 has been released with this bug fix, I'll close this.

closed 2020-11-06 by hifiberry