hifiberry/hifiberry-os #161: When updating to a new version;

Bertware, 2020-10-27:
**Describe the bug**
When updating from the previous version (202009dd) to the latest version (20201022) using the built-in updater, the `dtoverlay` parameter in `/boot/config.txt` was changed from `dtoverlay=hifiberry-dacplusadcpro` to `dtoverlay=hifiberry-dacpluspro`. As a result, the aux-in feature becomes unavailable until the user corrects the dtoverlay. After correcting the dtoverlay, the aux-in feature works again.

**HiFiBerryOS version**
20201022

**HiFiBerry sound card**
DAC + ADC Pro with AAMP60

**To Reproduce**
Steps to reproduce the behavior:
1. Use the built-in updater to update to 20201022. This is the first time that I used the built-in updater, so the issue may not be restricted to this specific build.

**Expected behavior**
The `dtoverlay` should be left untouched, or in case it uses auto-detect, it should properly detect dac+adc cards.

hifiberry, 2020-10-27:
Ok, looks like a detection bug. The config.txt file will usually not be touched, but it had to be changes this times as the kernel changed from 4.x to 5.x

Can you post the output of

/opt/hifiberry/bin/readhat

Bertware, 2020-10-27:
This is the output of the script, as well as the output of some files used by the script

```
# /opt/hifiberry/bin/readhat
/opt/hifiberry/bin/readhat: line 24: /sys/class/i2c-adapter/i2c-/new_device: No such file or directory
cat: /tmp/eeprom.txt: No such file or directory
cat: /tmp/eeprom.txt: No such file or directory
```
```
# ls /sys/class/i2c-adapter/
i2c-1
```
```
# ls /sys/class/i2c-adapter/i2c-1/
1-004a  1-004d  delete_device  device  i2c-dev  name  new_device  of_node  power  subsystem  uevent
```
```
# cat /proc/device-tree/hat/product
DAC+ ADC Pro
```

hifiberry, 2020-10-27:
Ok. Looks like this is still not working perfectly ;-)

hifiberry, 2020-10-28:
Some fixes have been implemented for this. They will be integrated in the next release.
https://github.com/hifiberry/hifiberry-os/commit/d1614ede99f9b1384daec605a55c9fac91dd674b

closed 2020-10-28 by hifiberry

Bertware, 2020-10-28:
Thanks for the quick fix!