hifiberry/hifiberry-os #162: Everything runs under the user root

jonasschnelli, 2020-10-28:
Is there a reason why all userspace applications like `/bin/python3 -m usagecollector.server` or `librespot` run under the root user?

I wished I could use the HIFIBerryOs as base for other applications but seeing that everything runs under root makes me not want to run other services on OS.

Is there a reason for that and how complicated would it be to change that?

hifiberry, 2020-10-28:
This is designed as an embedded system for audio playback only, not as a general purpose server that can other stuff. Therefore, it's a design choice to run everything as root. 
Adding service accounts for every application would complicate the whole environment quite a lot. We don't plan to do this in the near future.

closed 2020-10-28 by hifiberry

jonasschnelli, 2020-10-28:
Understood.
Though it wouldn't need service accounts for *every* application. I guess just a single user called `hifiberry` or so that runs all daemons, etc. to shield possible security incidents from directly root access.

hifiberry, 2020-10-28:
It's more complicated than this. We're open to contributions, but we don't plan to spend time on this in the near future as there are still a lot of interesting features on our list that have a higher priority for us.

jonasschnelli, 2020-10-28:
I can bet that it is not easy. Especially with a web-interface that has the capacity to stop and start deamons, etc.
Depending on the expected timeline/EoL of your product, I highly recommend to invest in a stable security concept... LAN appliances are not immune to attacks.