hifiberry/hifiberry-os #164: Include Python-smbus package for SMBus I2C support AudioControl2 extensions

Bertware, 2020-10-31:
**Is your feature request related to a problem? Please describe.**
I would like to develop some AudioControl extensions for showing metadata on various I2C/SPI led displays. These displays often have libraries available, but rely on SMBus for I2C communication. Adding this package to the buildroot project would allow developers to use these libraries in their AudioControl extensions.

Examples:
https://github.com/pimoroni/scroll-phat-hd/blob/master/library/scrollphathd/is31fl3731.py
https://github.com/pimoroni/microdot-phat/blob/master/library/microdotphat/matrix.py
https://github.com/dbrgn/RPLCD (supports Smbus2 as a fallback, see below)

**Describe the solution you'd like**
I would like to see the `python-smbus` added to the buildroot project

**Describe alternatives you've considered**
There is an Smbus2 library, written in pure python, which tries to be a drop-in replacement for Smbus. However, this library does not seem to work 100% when I modified the scrollphathd library.  Projects which already use Smbus2 only use it as a fallback, indicating that smbus is the better alternative when available.

**Additional context**
Since this is a widely-used library, it would allow many use-cases not described here. Any I2C hardware could be linked to AudioControl, such as port expanders, displays, I2C rotary encoders, ...

hifiberry, 2020-10-31:
Should be doable. When building you're own HiFiBerryOS version, you can simply select it in the configuration.

hifiberry, 2020-11-01:
You can try the latest experimental

/opt/hifiberry/bin/update --experimental

Bertware, 2020-11-01:
I tried the experimental branch, smbus works perfectly now!
Thanks for the quick fix!

closed 2020-11-01 by hifiberry