hifiberry/hifiberry-os #168: custom splash screen

stutteringp0et, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
Using the OS with an attached screen, it would be awesome to have the ability to customize the splash screen

**Describe the solution you'd like**
Ideally, I'd like to be able to place a correctly formatted and named image into one of the data directories, and have the boot sequence use that image instead of the default.

**Describe alternatives you've considered**
I've been reading about the process used to create a custom splash screen using psplash (the splash used in hifiberry-os)

**Additional context**
I'm building a portable music system using the beocreate, raspi-4, raspi-touchscreen, a 40v battery, and a Pyle subwoofer.  I gave the system a name (The Weapon), and would like the ability to make it boot up with that name on the splash screen.

hifiberry, 2020-11-03:
This would require to compile your own version as the splash screen is hard-coded and built in during compile time.

closed 2020-11-03 by hifiberry

stutteringp0et, 2020-11-03:
@hifiberry would you entertain a pull request to add this functionality?

hifiberry, 2020-11-03:
Depends on the pull. If you find a way NOT to put in your logo and make it configurable, we might think about it. However, we won't include other logos into our build.

stutteringp0et, 2020-11-03:
Absolutely, it would default to your current splash screen.

hifiberry, 2020-11-03:
Not sure how you want to do this. AFAIK psplash doesn't support this. What are you planning to do?

stutteringp0et, 2020-11-03:
Ubuntu uses plymouth, which allows user configuration, so I was going to look into that.

hifiberry, 2020-11-03:
Have a look! However, I'm not sure if this can display messages. That's an important feature that we don't want to loose.

stutteringp0et, 2020-11-03:
Plymouth can display messages.  I wouldn't want to remove any feature to add one.

With your blessing, I'll look at this more closely and come back with a pull request when I have something working.

hifiberry, 2020-11-03:
Looking forward to this.

Leshauts, 2023-11-07:
Would be nice to be able to custom (or access html files?) the "Now playing" screen. Like choose a uniform background or make the text of the artsit and the song a bit bigger.