hifiberry/hifiberry-os #169: squeezelite volume control not working

durgadas311, 2020-11-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When using the mouse on HDMI display, the volume control knob reacts to the mouse but cannot be moved. The only way to control the volume is from the Logitec Media Server app on the host, or from a tablet (smart phone). With an attached keyboard, UP/DOWN arrow keys do work. I am waiting for the remote control to arrive, so I suspect it will work as the keyboard did.

**HiFiBerryOS version**
20200908

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Connect display (HDMI) and mouse (USB)
2. Connect using squeezelite to Logitec Media server
3. click and drag volume control knob - it does not move.

**Expected behavior**
Should be able to drag volume control knob with mouse

**Screenshots**
none

**Browser (if applicable)**
N/A

**Additional context**
Add any other context about the problem here.

hifiberry, 2021-01-20:
Do you still see this on the latest release?

durgadas311, 2021-01-20:
This can be closed, for now. once I update I will re-check and if it still exists then will re-open or create a new issue.

closed 2021-01-20 by durgadas311