hifiberry/hifiberry-os #170: Make screen (HDMI) text/symbols configurable for larger displays

durgadas311, 2020-11-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
I have a 4K TV connected to the HiFiBerry and the text and controls are too small to be easily viewed. Oddly, the Album Cover image is "normal" (full size).

**Describe the solution you'd like**
I'd like to configure (or have HiFiBerry automatically) text and symbol sizes to reflect the screen size.

**Describe alternatives you've considered**
none

**Additional context**
none

hifiberry, 2020-11-05:
Can you post a picture how this looks like on your screen?

durgadas311, 2020-11-05:
[image]

Here is a photo of the screen, showing the disproportionate text/controls size.

hifiberry, 2020-11-05:
Ok, that looks really strange. Is this the latest 20201101 version? Because the size buttons look like an older bug that should have been fixed already

durgadas311, 2020-11-05:
This was 20200908. If the 20201101 version is stable, I can work through the update process on it, but if the fix should have already in 20200908 then I'll wait for the latest to stabilize.

durgadas311, 2020-11-05:
Again, this is a 4K TV.

hifiberry, 2020-11-05:
The fix won't change the font size, but only the buttons that are mis-aligned.