hifiberry/hifiberry-os #173: Remote control not working

durgadas311, 2020-11-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Got the remote control device recommended, but it appears to not generate any actions for any keys. Even volume control does not work. This is using squeezelite.

**HiFiBerryOS version**
20200908

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
Steps to reproduce the behavior:
1. connect remote control fob
2. press remote control buttons.
3. no action on HiFiBerry

**Expected behavior**
Remote control should at least affect volume, play, pause, next, previous

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
I initially plugged the remote fob into my PC, and confirmed that most buttons generated codes (home and top-level-menu did not seem to generate codes).

Does the HiFiBerry need to reboot to recognize the remote fob? My PC automatically added it.

durgadas311, 2020-11-05:
I also see that the remote does not have any obvious play/pause, next, previous buttons. Unclear how these functions map to the remote.

hifiberry, 2020-11-05:
Remote control needs to be plugged in before booting.

durgadas311, 2020-11-05:
OK, tried RESTART and it did not work. The unit seemed to be stuck after powering off, perhaps not able to restart/boot. I cycled power and booted, and the remote partially worked. up/down/left/right moved the mouse pointer. volume control worked. But I could not actually perform any actions - OK did not have any affect. Then, up/down/left/right stopped working altogether.

durgadas311, 2020-11-05:
The unit seems to be partially jammed now. Even the mouse cannot pause/play/next/previous. And the title/artist is not showing and the album cover does not display. I'll reboot again. It also does not automatically connect to squeezelite after reboot, I have to initiate from the server.

durgadas311, 2020-11-05:
power cycled and "normal" function restored, did not try remote control again. It would appear that maybe the "keyboard" module/thread got hung or crashed.

durgadas311, 2020-11-05:
I don't see anything useful in /var/log. dmesg only shows the current boot. I'll try and reproduce again and check dmesg for something. Is there any other log being kept?

durgadas311, 2020-11-05:
Remote is working differently now. Seems I need to understand exactly how to use it. Is there documentation on this remote? It seems to have two modes (at least). If I press the mouse pointer button, it toggles between either controlling the mouse pointer or some other functions that I don't understand.

hifiberry, 2020-11-05:
Please do not use the bug tracker for questions like this. Please post in the HiFiBerryOS forum with questions about the remote control:
https://support.hifiberry.com/hc/en-us

closed 2020-11-05 by hifiberry