hifiberry/hifiberry-os #174: Resuming radio after a long pause doesn't really work

schwachstelle, 2020-11-06:
**Describe the bug**
When pausing radio station for at least ten minutes, after resuming radio goes on for ~30 seconds an then stops.

**HiFiBerryOS version**
20201101

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. play a radio station
1. press pause on web gui or remote control
1. wait for at least 10 minutes
1. press pause again to resume playing the radio station
1. radio resumes playing for ~30 seconds and then stops.

**Expected behavior**
radio should not stop after resuming

**Screenshots**
no screenshot

**Browser (if applicable)**
happens with remote control as well

**Additional context**
I suppose, when pressing pause HiFiBerryOS keeps on streaming the radio station until some buffer is filled. When resuming it plays this buffer and stops afterwards.
The question is, what to expect when pausing a radio station. If you really want to 'pause', you have to keep on streaming while pausing, even if pausing for hours. But I think for playing radio it would be more sensible to implement pause/resume as stop/start again.

hifiberry, 2020-11-06:
This is the expected behavior as a there is no "pause" for a radio station. 
Best thing we might be able to do is to deactivate the "play" button. To restart a radio station, you have to re-select it in the radio stations menu.

schwachstelle, 2020-11-06:
But this would make it impossible to resume a radio station with the remote control. I would really like to have this possibility. Right now it only works if the pause is not too long.

hifiberry, 2020-11-06:
That right, but that's how the system works. Feel free to provide a patch that implements this.

schwachstelle, 2020-11-07:
Ok, if I find some time, I'll have a look into it. Can you give me two pointers in context with the radio player?
* where is the URL handed over
* how is the pause command signaled to the player

hifiberry, 2020-11-07:
Play/Pause is handled by Audiocontrol:
https://github.com/hifiberry/audiocontrol2/blob/master/ac2/players/mpdcontrol.py

Starting a radio station is done via start-radio:
https://github.com/hifiberry/hifiberry-os/blob/development/buildroot/package/webradio/start-radio

hifiberry, 2020-11-07:
You probably have to check the mpd queue for radio URLs and handle these differently than local files

schwachstelle, 2020-11-07:
ok, thanks!

closed 2024-01-12 by hifiberry