hifiberry/hifiberry-os #179: 'bad address' when trying to ping local hostnames

Bertware, 2020-11-14:
**Describe the bug**

When using the ping command for local hostnames, a 'bad address' error is thrown. When trying to resolve the same hostname through nslookup, it resolves correctly. While this might seem insignificant, it might be a symptom of underlying network issues.

I pinged google successfully:
```
# ping google.com
PING google.com (216.58.211.142): 56 data bytes
64 bytes from 216.58.211.142: seq=0 ttl=118 time=2.532 ms
64 bytes from 216.58.211.142: seq=1 ttl=118 time=5.998 ms
^C
--- google.com ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 2.532/4.265/5.998 ms
```

Then I tried to ping a raspberry pi by its local hostname, but this failed. nslookup worked. pinging the ip directly worked.
```
# ping rpi-01.local
ping: bad address 'rpi-01.local'
# nslookup rpi-01.local
Server:         192.168.1.2
Address:        192.168.1.2:53

Name:   rpi-01.local
Address: 192.168.1.2

Non-authoritative answer:

# ping -c 2 192.168.1.2
PING 192.168.1.2 (192.168.1.2): 56 data bytes
64 bytes from 192.168.1.2: seq=0 ttl=64 time=2.267 ms
64 bytes from 192.168.1.2: seq=1 ttl=64 time=5.659 ms

--- 192.168.1.2 ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 2.267/3.963/5.659 ms
```

Same with another pi: ping failed to resolve the hostname, nslookup worked. pinging the ip directly worked (with timeouts since that pi was offline atm).
```
# ping rpi-02.local
ping: bad address 'rpi-02.local'
# nslookup rpi-02.local
Server:         192.168.1.2
Address:        192.168.1.2:53

Name:   rpi-02.local
Address: 192.168.1.3

Non-authoritative answer:
```

The reason why I'm filing this bug is that I discovered this while working on the snapcast implementation. I'm determining the snapserver address through zeroconf, I obtain the server address successfully, but python fails to communicate even though the requests work perfectly in postman. The fact that even ping has issues, and that everything works with the IP, suggests this is a bug in the operating system.

This stackoverflow thread discusses how this could be related to zeroconf/avahi not working properly: https://unix.stackexchange.com/questions/289930/cannot-ping-nor-curl-a-local-domain-but-nslookup-works

This stackoverflow thread discusses the opposite problem of nslookup failing while ping works: https://unix.stackexchange.com/questions/124283/busybox-ping-ip-works-but-hostname-nslookup-fails-with-bad-address

**HiFiBerryOS version**
20201101

**To Reproduce**
Ping a local host by it's hostname. To reproduce the issue that caused me to start investigating in exactly the same way, run a service such as snapserver which advertises its name using zeroconf, and obtain the address using the following code snippet:

```
def get_zeroconf_server_address():
    zerocfg = Zeroconf()
    service_info = zerocfg.get_service_info("_snapcast._tcp.local.", "Snapcast._snapcast._tcp.local.", 3000)
    if service_info is None:
        logging.error("Failed to obtain snapserver address through zeroconf!")
        return None
    logging.info("Obtained snapserver address through zeroconf: " + service_info.server)
    # Remove the trailing dot, as it is invalid in a host with port number
    return service_info.server.rstrip(".")
```
Then try to make a call to that address. It will fail with the following error:
```
requests.exceptions.ConnectionError: HTTPConnectionPool(host='rpi-01.local', port=1780): Max retries exceeded with url: /jsonrpc (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb486ada8>: Failed to establish a new connection: [Errno
-2] Name or service not known'))
```
Again, name or service not known points to DNS issues in the operating system, possibly caused by avahi or some other zeroconf daemon that might be missing or misbehaving. Other operating systems and devices resolve the name without problems, including another pi in the network running Raspbian and a Windows computer.

**Expected behavior**
Programs such as ping, and the python requests module, should have no problem resolving local dns names obtained through zeroconf.

**Additional context**
While I could work around this, I believe it would be good to fix this one, as it could lead to other unpredictable behaviour in other music players (such as snapclient etc)

hifiberry, 2020-11-15:
mDNS is not DNS (is's similar, but quite different) and not supported on OS level. An application that required mDNS (or also CIFS) name resolution has to integrate these directly.
Therefore, using Zeroconf name resolution in your application is the correct way to implement this.

imgrant, 2020-11-15:
I don't think that is strictly correct, is it? HiFiBerryOS is using `systemd-resolved` for name resolution, which supports using mDNS for name resolution. `systemd-resolved` usually enables this by default, but it also needs to be enabled per connection in `systemd-networkd`. You can set this ephemerally to verify, for example:

```bash
# ping server.local
ping: bad address 'server.local'
# systemd-resolve --set-mdns=yes --interface=eth0
# ping server.local
PING server.local (192.168.139.100): 56 data bytes
64 bytes from 192.168.139.100: seq=0 ttl=64 time=1.242 ms
64 bytes from 192.168.139.100: seq=1 ttl=64 time=0.570 ms
^C
--- server.local ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 0.570/0.906/1.242 ms
```

HiFiBerryOS also has `avahi` installed, which could cause conflicts if they both serve mDNS resolution, but it doesn't look like HiFiBerryOS is configured to use Avahi for name resolution so it should be okay. Worth testing though.

Bertware, 2020-11-15:
@imgrant That worked! 

I'm using wireless on my test setup so had to run with wlan0, but I guess it can just be enabled on any interface.
```
# systemd-resolve --set-mdns=yes --interface=wlan0
# ping rpi-01.local
PING rpi-01.local (192.168.1.2): 56 data bytes
64 bytes from 192.168.1.2: seq=0 ttl=64 time=3.724 ms
64 bytes from 192.168.1.2: seq=1 ttl=64 time=5.123 ms
```

As soon as I ran this command, my snapcastMPRIS implementation (which was stuck in a reboot loop since it couldn't connect to snapserver) started working as well. 

@hifiberry Any chance this command/config could be included in the buildroot?

hifiberry, 2020-11-15:
While we can include it, I would strongly recommend to implement mDNS name resolution in your code - especially as it seems you're already using Zeroconf already for service discovery. You simply can't rely on a system having an mDNS resolver installed.

imgrant, 2020-11-15:
@Bertware To enable it persistently, set `MulticastDNS=resolve` or `MulticastDNS=yes` in the systemd network unit file (in the `[Network]` section) for the interface(s) — see https://www.freedesktop.org/software/systemd/man/systemd.network.html
On my Raspberry Pi HiFiBerryOS install, the network files are `/etc/systemd/network/wireless.network` and `/etc/systemd/network/eth0.network`.

@hifiberry I don't quite agree (unless there is something I am missing). Whilst implementing mDNS name resolution within the code might arguably be a more robust solution, at the same time, it's less flexible and IMO it's better to leverage existing solutions than re-invent the wheel. In the case of HiFiBerryOS, it is a self-contained OS/system, so if mDNS is enabled in HiFiBerryOS, I would say you absolutely can rely on it.

hifiberry, 2020-11-26:
@imgrant I understand your reasoning. However, audiocontrol is not fully tied to HiFiBerryOS, but can also be used standalone. Therefore, I still think that it's best to implement this within audiocontrol. 
However, enabling mDNS in HiFiBerryOS might still be useful for other things.

hifiberry, 2020-11-26:
There is now a script that can enable mDNS lookups:
https://github.com/hifiberry/hifiberry-os/commit/8c786899399e31cede8b67028b1fb94057aea667

They are still not enabled by default

imgrant, 2020-11-26:
👍 Ah yes, I forget bits like audiocontrol have other lives!

hifiberry, 2020-12-24:
As this is now using mDNS, I'll close this feature request.

closed 2020-12-24 by hifiberry