hifiberry/hifiberry-os #182: SnapcastMpris requirements not installed

Bertware, 2020-11-28:
**Describe the bug**
The SnapcastMpris project contains a requirements.txt file, but these dependencies are not installed automatically. As a result, snapcast fails to start.

Running  `pip install -r /opt/snapcastmpris/requirements.txt` installs the requirements and fixes this bug

**HiFiBerryOS version**
20201123

**To Reproduce**
Install the experimental build, try to enable snapcast.

**Expected behavior**
Requirements are installed (either when building the image or on first boot)

hifiberry, 2020-11-29:
That's normal. requirements need to be packaged separately there is no automatic installation. What package(s) specifically is/are missing?

hifiberry, 2020-11-29:
OK, looks like you added several dependencies here. Luckily these are all available as packages in buildroot already. They should be included in the next experimental build.
Another thing: Please have a look at this bug: https://github.com/hifiberry/snapcastmpris/issues/3

hifiberry, 2020-11-30:
The requirements have now been integrated.  However, I reverted the snapcastmpris to the previous version for now until the other bug is fixed.

closed 2020-11-30 by hifiberry