hifiberry/hifiberry-os #183: mpd-mpris was apparently removed, no more MPD?

cjeanner, 2020-11-30:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
With latest Pi3 image (hifiberryos-20201101-pi3.img), the mpd-mpris was apparently removed, while systemd still knows about it.
It seems no more mpd service is present, or at least started, leading to a constant error log such as:
```
Nov 30 07:18:43 hifiberry upmpdcli[407]: :2:src/mpdcli.cxx:141::MPDCli::openconn failed: Connection refused
Nov 30 07:18:43 hifiberry upmpdcli[407]: :2:src/main.cxx:661::MPD connection failed
Nov 30 07:18:44 hifiberry ympd[223]: MPD connection failed.
Nov 30 07:18:45 hifiberry ympd[223]: MPD connection: Connection refused
Nov 30 07:18:46 hifiberry ympd[223]: MPD connection failed.
Nov 30 07:18:46 hifiberry python3[299]: WARNING:root:mpd failed, trying to restart
Nov 30 07:18:46 hifiberry python3[1079]: Failed to restart mpd-mpris.service: Unit mpd-mpris.service not found.
Nov 30 07:18:47 hifiberry ympd[223]: MPD connection: Connection refused
Nov 30 07:18:47 hifiberry mpd-update-notifier[1080]: MPD error: Connection refused
Nov 30 07:18:48 hifiberry ympd[223]: MPD connection failed.
Nov 30 07:18:49 hifiberry ympd[223]: MPD connection: Connection refused
Nov 30 07:18:49 hifiberry python3[299]: WARNING:root:mpd failed, trying to restart
Nov 30 07:18:49 hifiberry python3[1082]: Failed to restart mpd-mpris.service: Unit mpd-mpris.service not found.
Nov 30 07:18:50 hifiberry ympd[223]: MPD connection failed.
Nov 30 07:18:51 hifiberry ympd[223]: MPD connection: Connection refused
```

**HiFiBerryOS version**
20201101-pi3
(taken from the image name - the "about" doesn't work in the interface)

**HiFiBerry sound card**
Not relevant - and I don't remember what I've put in the box. If this is really needed, lemme know - but I really doubt it has anything to do with the current issue

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest image
2. Try to get MPD up'n'running
3. You can connect to SSH in parallel and run ```journalctl -f```
4. See error

**Expected behavior**
If mpd-mpris is rightfully removed, its dependencies and configurations and services shouldn't be there anymore.

**Screenshots**
None

**Browser (if applicable)**
None

**Additional context**
I've tracked back the removal of mpd-mpris here: https://github.com/hifiberry/hifiberry-os/commit/193ccb9054d0d7673039ffe0678d41af01237ee7

hifiberry, 2020-11-30:
MPD is still part of HiFiBerryOS, only mpd-mpris has been removed as the connection to mpd is handled differently.
I gues,s there's another issue here.

What's the output of

journalctl -u mpd

cjeanner, 2020-11-30:
Hello,
Thank you for the fast answer. Here's what I get with ```journalctl -u mpd```
```
-- Logs begin at Thu 2020-07-30 19:02:37 UTC, end at Mon 2020-11-30 08:04:00 UTC. --
Jul 30 19:02:41 hifiberry systemd[1]: Dependency failed for mpd.
Jul 30 19:02:41 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Jul 30 19:02:41 hifiberry systemd[1]: Dependency failed for mpd.
Jul 30 19:02:41 hifiberry systemd[1]: mpd.service: Job mpd.service/start failed with result 'dependency'.
Nov 30 08:01:45 hifiberry systemd[1]: Starting mpd...
Nov 30 08:01:45 hifiberry systemd[1]: mpd.service: Found left-over process 618 (update-mpd-db) in control group while starting unit. Ignoring.
Nov 30 08:01:45 hifiberry systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Nov 30 08:01:45 hifiberry systemd[1]: mpd.service: Found left-over process 638 (update-covers) in control group while starting unit. Ignoring.
Nov 30 08:01:45 hifiberry systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Nov 30 08:01:45 hifiberry systemd[1]: mpd.service: Found left-over process 639 (mpc) in control group while starting unit. Ignoring.
Nov 30 08:01:45 hifiberry systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Nov 30 08:01:45 hifiberry systemd[1]: Started mpd.
Nov 30 08:01:47 hifiberry mount-smb.sh[641]: INFO: root - return code 0 only if new covers were found
Nov 30 08:01:47 hifiberry mount-smb.sh[641]: INFO: root - loaded processed files from /data/library/music/getcovers.log
Nov 30 08:01:47 hifiberry mount-smb.sh[641]: INFO: root - Extracting covers from /data/library/music
Nov 30 08:01:47 hifiberry mount-smb.sh[641]: INFO: root - saved processed files to /data/library/music/getcovers.log
Nov 30 08:01:47 hifiberry mount-smb.sh[641]: INFO: root - Stats: {}
Nov 30 08:01:47 hifiberry mount-smb.sh[638]: Looking up images ...
Nov 30 08:01:47 hifiberry mpd[640]: exception: Input plugin 'qobuz' is not configured: No Qobuz app_id configured
Nov 30 08:01:47 hifiberry libsmbclient[640]: zeroconf: No global port, disabling zeroconf
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [0] opened from local
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [1] opened from 127.0.0.1:50070
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [2] opened from local
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [3] opened from 127.0.0.1:50072
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [1] closed
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [4] opened from 127.0.0.1:50082
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [3] closed
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [5] opened from 127.0.0.1:50086
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [4] closed
Nov 30 08:01:47 hifiberry mount-smb.sh[639]: Updating DB (#1) ...
Nov 30 08:01:47 hifiberry mount-smb.sh[639]: volume:100%   repeat: off   random: off   single: off   consume: off
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [6] opened from 127.0.0.1:50088
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [5] closed
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [7] opened from 127.0.0.1:50090
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [0] closed
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [6] closed
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [8] opened from 127.0.0.1:50138
Nov 30 08:01:47 hifiberry libsmbclient[640]: client: [7] closed
Nov 30 08:01:48 hifiberry libsmbclient[640]: client: [9] opened from 127.0.0.1:50142
Nov 30 08:01:50 hifiberry libsmbclient[640]: client: [10] opened from 127.0.0.1:50144
Nov 30 08:02:01 hifiberry libsmbclient[640]: client: [11] opened from local
Nov 30 08:02:06 hifiberry mount-smb.sh[679]: INFO: root - retrieving covers from web services
Nov 30 08:02:06 hifiberry mount-smb.sh[679]: INFO: root - retrieving artist pictures from web services
Nov 30 08:02:06 hifiberry mount-smb.sh[679]: INFO: root - return code 0 only if new covers were found
Nov 30 08:02:06 hifiberry mount-smb.sh[679]: INFO: root - Extracting album information from /data/library/music
Nov 30 08:02:06 hifiberry mount-smb.sh[679]: INFO: root - processing /data/library/music/local
Nov 30 08:03:50 hifiberry libsmbclient[640]: client: [12] opened from local
Nov 30 08:03:50 hifiberry libsmbclient[640]: client: [12] closed
Nov 30 08:03:50 hifiberry libsmbclient[640]: update: added DiskStation-Music-zTuIu/[...]
```
It seems to work better after a fresh re-install (note: the previous log came also from a fresh install).

Here's also the output for ```journalctl -u ympd``` - though that service is maybe not relevant anymore?
```
Jul 30 19:02:39 hifiberry systemd[1]: Starting ympd MPD web interface...
Jul 30 19:02:39 hifiberry systemd[1]: Started ympd MPD web interface.
Jul 30 19:02:43 hifiberry ympd[227]: MPD connection: Timeout
Jul 30 19:02:43 hifiberry ympd[227]: MPD connection failed.
Jul 30 19:02:47 hifiberry ympd[227]: MPD connection: Timeout
Jul 30 19:02:47 hifiberry ympd[227]: MPD connection failed.
Jul 30 19:02:51 hifiberry ympd[227]: MPD connection: Timeout
Jul 30 19:02:51 hifiberry ympd[227]: MPD connection failed.
Nov 30 08:01:11 hifiberry ympd[227]: MPD connection: Timeout
Nov 30 08:01:11 hifiberry ympd[227]: MPD connection failed.
[... cut ...]
Nov 30 08:01:39 hifiberry ympd[227]: MPD connection failed.
Nov 30 08:01:43 hifiberry ympd[227]: MPD connection: Timeout while connecting
Nov 30 08:01:43 hifiberry ympd[227]: MPD connection failed.
Nov 30 08:01:47 hifiberry ympd[227]: MPD connection: Timeout while connecting
Nov 30 08:01:47 hifiberry ympd[227]: MPD connection failed.
Nov 30 08:01:48 hifiberry ympd[227]: MPD connected.
```

Last line didn't show up with the previous test.... Sounds weird, same way of installation :/. Maybe something's flappy in there?

hifiberry, 2020-11-30:
Hard to say now what might have been the issue before. But good to hear that it's working now.

closed 2020-11-30 by hifiberry

qerub, 2020-11-30:
I found this issue after doing a clean install of 20201101-pi3 yesterday and having some trouble. After some digging around I found at least one bug, namely leftover references to `mpd-mpris.service`:

```
# grep -r mpd-mpris /opt
/opt/beocreate/beo-extensions/mpd/index.js:		exec("systemctl enable --now mpd.service mpd-mpris.service ympd.service").on('exit', function(code) {
/opt/beocreate/beo-extensions/mpd/index.js:		exec("systemctl disable --now mpd.service mpd-mpris.service ympd.service").on('exit', function(code) {
/opt/hifiberry/etc/services:mpd-mpris.service
/opt/hifiberry/bin/start-radio:systemctl start mpd mpd-mpris
/opt/hifiberry/bin/check-system:for i in audiocontrol2 beocreate2 bluetooth bluealsa bluealsa-aplay dlnampris mpd mpd-mpris pigpiod raat shairport-sync sigmatcp snapcastmpris spotify squeezelite sshd ympd weston cog; do
```

hifiberry, 2020-11-30:
This has been fixed already in the development tree and the fix will be included in the next release.