hifiberry/hifiberry-os #188: Arranging sources doesn't work

imgrant, 2020-12-02:
**Description**
On the Sources page, clicking 'Arrange sources...' button enters a mode where the sources can be dragged up and down, to reorder them. When dragging a source, as soon as the mouse button is released, it moves back to its original place, and no-reordering takes place.

**HiFiBerryOS version**
20201202

**HiFiBerry sound card**
DAC+

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the 'Sources' tab/page
2. Click on 'Arrange sources...'
3. Drag one of the sources to a different position in the list
4. Observe that when you let go of the dragged source, it moves back to its original position and the list remains in the original order

**Expected behavior**
When releasing the dragged item in step 4 above, the list should re-order to reflect the desired position

**Browser**
 - OS: Windows 10
 - Browser: Chrome, Edge
 - Version: 87.0.4280.66 (Chrome), 87.0.664.52 (Edge)

hifiberry, 2020-12-02:
I can't reproduce it here. Please also check the Javascript console for error as in https://github.com/hifiberry/hifiberry-os/issues/189

imgrant, 2020-12-02:
Hmm, I don't see any errors, but might not be looking in the right place. Here's the output in the console (apologies, it doesn't paste well compared to viewing in-browser 😞), after: loading the page, clicking 'Arrange sources...', dragging & dropping one of them:

```vue.js:9055 Download the Vue Devtools extension for a better development experience:
https://github.com/vuejs/vue-devtools
vue.js:9064 You are running Vue in development mode.
Make sure to turn on production mode when deploying for production.
See more tips at https://vuejs.org/guide/deployment.html
beo-drag.js:187 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Beodrag @ beo-drag.js:187
(anonymous) @ interact-client.js:554
(anonymous) @ interact-client.js:809
beo-ui.js:184 Preparing menus...
beo-ui.js:387 Menus ready.
beo-ui.js:593 Setting appearance to light.
beo-comms.js:60 Connecting to living-room-hifi.lan...
(index):1 [DOM] Input elements should have autocomplete attributes (suggested: "username"): (More info: https://goo.gl/9p2vKq) <input type=​"text" name=​"text-input" value id=​"text-input-plain" autocorrect=​"off" autocapitalize=​"off" placeholder=​"Placeholder">​
beo-comms.js:83 Succesfully connected to living-room-hifi.lan.
beo-comms.js:172 now-playing useExternalArtwork undefined
beo-comms.js:148 {header: "useExternalArtwork", target: "now-playing", content: {…}}
beo-comms.js:148 {header: "basicProductInformation", target: "product-information", content: {…}}
beo-comms.js:148 {header: "setupStatus", target: "setup", content: {…}}
beo-comms.js:172 general activatedExtension {extension: "sources", deepMenu: null}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-comms.js:148 {header: "systemVolume", target: "sound", content: {…}}
beo-comms.js:148 {header: "settings", target: "ui", content: {…}}
onloadwff.js:71 [Violation] 'setTimeout' handler took 67ms
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-drag.js:187 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Beodrag @ beo-drag.js:187
toggleArrange @ sources-client.js:191
onclick @ (index):2469
sources-client.js:215 (6) [null, null, null, "music", null, null]0: null1: null2: null3: "music"4: null5: nulllength: 6__proto__: Array(0)
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}content: {sources: {…}, currentSource: null, focusedSource: null, sourceOrder: Array(6)}header: "sources"target: "sources"__proto__: Object
```

tuomashamalainen, 2020-12-03:
Hi Ian and thanks for reporting this issue. It's probably related to the other problem that has been mentioned.

I'm seeing a bunch of `null` in the second to last line – that's the array of arranged sources that's sent from the speaker to the UI. It shouldn't have nulls in it and that's probably causing the arrangement to not work properly. I assume the list of sources you're seeing in the interface is the same as in the other issue? There seems to be a mismatch between the internal source list and what you see.

Can you drill down to the message on the last line (or similar – header should be "sources")? Particularly I'm interested in the "sources" object under "content" and the "sourceOrder" array. 

Best,
Tuomas from Bang & Olufsen

imgrant, 2020-12-03:
@tuomashamalainen 
Here's an expansion of one of the objects when the page has loaded:

[image]
`sources` is populated, but `sourceOrder` is replete with nulls

When I click 'Arrange sources...', the yellow [Violation] message in the screenshot below appears, and then when I drag and drop a source, the final message showing the array with "music" and nulls appears:

[image]

(Apologies, never quite sure where to file these issues, maybe it's more BeoCreate than HiFiBerryOS?)

tuomashamalainen, 2020-12-03:
Thanks for this, Ian.

The yellow message is caused by the dragging function but should otherwise be unrelated as far as I can tell.

The source order is stored on the speaker in `/etc/beocreate/sources.json`. If you have SSH/SFTP access, can you check this file to see if all the nulls are also stored there?

For further testing, I would go ahead and delete that file and then restart the Beocreate server (the file will be automatically recreated):
`systemctl restart beocreate2`

When the server has restarted and you have reloaded the page, are you still unable to rearrange sources and do you still get these nulls in the source order?

Many thanks!

> (Apologies, never quite sure where to file these issues, maybe it's more BeoCreate than HiFiBerryOS?)

Generally any issue in the user interface (like this one) is Beocreate and other underlying stuff is with the core OS, but don't beat yourself up about it as the line is often fuzzy. We do monitor across both repositories so we should be able to react where necessary.

Best,
Tuomas

imgrant, 2020-12-03:
Yes, deleting the `sources.json` file and restarting `beocreate2` fixed it, thanks!
It also fixed #189, I will comment & close there.

closed 2020-12-03 by imgrant

tuomashamalainen, 2020-12-03:
Good to hear, thanks!

I will run some more experiments here to try and replicate the problem so that I can add precautions in case it might occur again. No one should have to SSH in to fix anything 😉

Best,
Tuomas from Bang & Olufsen