hifiberry/hifiberry-os #189: Disabled sources aren't shown separately

imgrant, 2020-12-02:
**Description**
On the sources page, underneath the list of (active) sources, is a fold-out menu titled "Disabled sources", but it is always empty, even if some of the sources are disabled. All sources, enabled and disabled are shown in a single list above.

**HiFiBerryOS version**
20201202

**HiFiBerry sound card**
DAC+

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Sources'
2. Click on any source and disabled it by unticking the 'On' check circle
3. Return to the sources page
4. Observe that all the sources, enabled and disabled, are shown in a single list
5. Click the grey, right-pointing caret before the 'Disabled sources' header, underneath the list of sources and above the 'Arrange sources...' button
6. Observe the caret rotates to point downwards, indicating the list is folded out, but that it is empty; no disabled sources are shown here (as above, they are mixed with the list of enabled sources)

**Expected behavior**
Any disabled sources would be shown separately, in the 'Disabled sources' foldout menu, underneath the main list, which should show only enabled sources

**Screenshots**
In the image below, several sources (Squeezelite, Spotify Connect) are disabled, yet they appear in the primary list; the disabled sources list below is empty.
[image]

**Browser (if applicable)**
 - OS: Windows 10
 - Browser: Chrome, Edge
 - Version: 87.0.4280.66 (Chrome), 87.0.664.52 (Edge)

**Additional context**
Add any other context about the problem here.

hifiberry, 2020-12-02:
Interesting behaviour. Unfortunately I can't reproduce it here. Can you check your Javascript console in the browser for errors?

imgrant, 2020-12-02:
No errors as far as I can see. Output is similar to that in #188, but for completeness' sake, here is the console output after: loading the page, toggling the disabled sources foldout caret, going into Squeezelite, enabling it, coming back out, then going back in, disabling it and coming back out to the sources page again. Nothing shows in the disabled sources list.

```vue.js:9055 Download the Vue Devtools extension for a better development experience:
https://github.com/vuejs/vue-devtools
vue.js:9064 You are running Vue in development mode.
Make sure to turn on production mode when deploying for production.
See more tips at https://vuejs.org/guide/deployment.html
beo-drag.js:187 [Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Beodrag @ beo-drag.js:187
(anonymous) @ interact-client.js:554
(anonymous) @ interact-client.js:809
beo-ui.js:184 Preparing menus...
beo-ui.js:387 Menus ready.
beo-ui.js:593 Setting appearance to light.
beo-comms.js:60 Connecting to living-room-hifi.lan...
(index):1 [DOM] Input elements should have autocomplete attributes (suggested: "username"): (More info: https://goo.gl/9p2vKq) <input type=​"text" name=​"text-input" value id=​"text-input-plain" autocorrect=​"off" autocapitalize=​"off" placeholder=​"Placeholder">​
beo-comms.js:83 Succesfully connected to living-room-hifi.lan.
beo-comms.js:172 now-playing useExternalArtwork undefined
beo-comms.js:148 {header: "useExternalArtwork", target: "now-playing", content: {…}}
beo-comms.js:148 {header: "basicProductInformation", target: "product-information", content: {…}}
beo-comms.js:148 {header: "setupStatus", target: "setup", content: {…}}
beo-comms.js:172 general activatedExtension {extension: "sources", deepMenu: null}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-comms.js:148 {header: "systemVolume", target: "sound", content: {…}}
beo-comms.js:148 {header: "settings", target: "ui", content: {…}}
onloadwff.js:71 [Violation] 'setTimeout' handler took 80ms
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}content: {sources: {…}, currentSource: null, focusedSource: null}header: "sources"target: "sources"__proto__: Object
beo-comms.js:172 general activatedExtension {extension: "squeezelite", deepMenu: null}
beo-comms.js:148 {header: "squeezeliteSettings", target: "squeezelite", content: {…}}
beo-comms.js:148 {header: "squeezeliteSettings", target: "squeezelite", content: {…}}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-comms.js:172 general activatedExtension {extension: "sources", deepMenu: null}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-comms.js:172 general activatedExtension {extension: "squeezelite", deepMenu: null}
beo-comms.js:148 {header: "squeezeliteSettings", target: "squeezelite", content: {…}}
beo-comms.js:148 {header: "squeezeliteSettings", target: "squeezelite", content: {…}}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
beo-comms.js:172 general activatedExtension {extension: "sources", deepMenu: null}
beo-comms.js:148 {header: "sources", target: "sources", content: {…}}
```

imgrant, 2020-12-03:
As detailed in #188, the fix for this is:
1. Delete `/etc/sources/beocreate.json`
2. Restart the UI with `systemctl restart beocreate2`

Thanks @tuomashamalainen!

closed 2020-12-03 by imgrant