hifiberry/hifiberry-os #190: Could not Boot with RPI 2B+

Fabibo, 2020-12-03:
Hello,

Precondition:
I have an old RPI 2B+ [[2 Mod. B v1.2]](https://de.wikipedia.org/wiki/Raspberry_Pi) (architecture is nearly to the RPI 3).
Ontop is a Hifiberry Amp2 connected.
Power (12 V) is directly conntected at the Amp2.
Normal Raspbian has no problem to boot.

Current State:
After Installation of HifiBerryOS on Card, i cant boot the system.
The green LED pop ups 2 Times and over HDMI i can just see the PI colored Error Screen.

Anyone able to boot HifiBerryOS on a RPI 2b+? 

Thanks so much!
Fabian

hifiberry, 2020-12-04:
What version did you try? Did you also try an older release? 
https://github.com/hifiberry/hifiberry-os/releases

Fabibo, 2020-12-04:
Hi,

Yes I already tried old Versions. An now I also tried the new one.
I also tried to put [bcm2710-rpi-2-b.dtb](https://github.com/raspberrypi/firmware/tree/master/boot) because in the RPI 2 Versions there is only the bcm2709 File.

But always the same procedure.. No boot and directly the rainbow screen.
Somebody already tried to boot HifiberryOS on an pi 2 1.2?
Maybe i should buy a new pi4 ;)

Fabibo, 2020-12-10:
I now just bought an Rpi 4. Working like a charm.

closed 2020-12-10 by Fabibo