hifiberry/hifiberry-os #191: allow passwords to be viewed when entered to check them

cwgamma, 2020-12-12:
Most password entry fields permit you to check what you have typed as a password by clicking on a little eye icon to the left of the entry field to make the password visible. That would be much appreciated in the HiFiBerry web app

tobiv, 2021-01-06:
I should be able to add this. 
@hifiberry Is it possible to only build/run the UI to test this or is it necessary to build the entire OS?

hifiberry, 2021-01-07:
@tobiv You can't easily run the UI without HiFiBerryOS, but for development, you don't need to build the whole system, but only change the UI and copy it over:
https://github.com/bang-olufsen/create

tobiv, 2021-01-26:
I created a pull request at https://github.com/bang-olufsen/create/pull/134

tobiv, 2021-01-26:
Pull request was merged; issue can be closed.
(Note that you'll have to force reload the page (`Ctrl-Shift-R` in most browsers) for this to take effect, once it has been released of course.)

closed 2022-07-21 by hifiberry