hifiberry/hifiberry-os #192: Rogue DHCP Server

bayoda, 2020-12-13:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
DHCPd Rogue Server in the Network on WIFI.  No GUI Setting to disable that. 

**HiFiBerryOS version**
20201202

**HiFiBerry sound card**
hifiBerry AMP2 

**To Reproduce**
Steps to reproduce the behavior:
No mather if fixed or not the  DHCPD is running 

**Expected behavior**
this shouldn't happen if it is not configured as AccessPoint

**Screenshots**

**Browser (if applicable)**
Linux / ipad OS 
Chrome / Firefox / Safari - seen on the firewall and the unifi switch

**Additional context**
Pls give us a possibility to turn of the AP Feature or the DHCP Demon.

greetings 

Michael

hifiberry, 2020-12-13:
See 
https://github.com/hifiberry/hifiberry-os/issues/109#issuecomment-661689544

closed 2020-12-21 by hifiberry

wjdp, 2021-05-17:
@hifiberry just had a similar issue. Hifiberry was advertising DHCP leases when in wireless client mode. Noticed when my wired desktop (same subnet) couldn't get online. Have unplugged the Pi for now to restore my network.

wjdp, 2023-05-10:
This happened today again on my network. Took a while to find as most devices picked up the correct addresses, only a few got bad ones. Is there a plan to disable DHCP when not acting as an AP?

hifiberry, 2023-05-11:
DHCP is disabled when not acting as an AP