hifiberry/hifiberry-os #194: Hifiberry OS builds with libupnp instead of libupnp18

Nazdravi, 2020-12-14:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When compiling hifiberry-os the package "libupnp" is selected, but libupnp18 provides version >= 1.8
../buildroot-4/build/hifiberry-mpd-0.22.2/src/lib/upnp/meson.build:1:0: ERROR: Invalid version of dependency, need 'libupnp' ['>= 1.8'] found '1.6.25'.

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD
github sources "master" from december 14th, 2020
buildroot-2020.11
target: raspi4

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
./build-config 4
./config
./compile

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Regards,
Nazdravi

hifiberry, 2020-12-14:
I guess, you're using an unpatched buildroot version. HiFiBerryOS requires some patches (including libupnp). Therefore, you should retrieve the correct buildroot version using 

./get-buildroot

Nazdravi, 2020-12-15:
Thanks. Solved.

closed 2020-12-15 by hifiberry