hifiberry/hifiberry-os #196: Saved PSK gets truncated after a few days and wifi connection fails

jeLee6gi, 2020-12-16:
**Describe the bug**
After a few days, wifi randomly disassociates from my AP or doesn't reconnect after reboot. According to AP logs, it attempts to associate with an invalid PSK. Updating the PSK in the web interface solves the problem temporarily.

The last time it occured, I checked `/etc/wpa_supplicant.conf` and noticed that my password was truncated to the first eleven characters. The full password was `"VhqO(J9MU:~=z0{Yf<DK(xg1`.

```
country=DE
ap_scan=1

network={
	ssid="XXX"
	psk=""VhqO(J9MU:"
}
```

**HiFiBerryOS version**
20201101

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
1. Connect to wifi using PSK `"VhqO(J9MU:~=z0{Yf<DK(xg1`
2. Wait a few days

**Expected behavior**
The PSK doesn't get truncated

**Screenshots**

**Browser (if applicable)**

**Additional context**
I suspected that the double quotation marks in my password could be a problem and changed my AP password to one that doesn't use them. Time will tell if that helped, but I don't understand why the password gets truncated in the first place.

tuomashamalainen, 2020-12-16:
Hey there,
thanks for reporting this. I will try to reproduce this fault and fix it. There's probably a bug in the code that deals with the keys when configuring networks.

Best,
Tuomas from Bang & Olufsen

jeLee6gi, 2020-12-16:
Thank you for looking into this, I had a thought about why it gets truncated the way it does. The quotation marks in my password probably end the value for `psk` and the colon gets interpreted as the end of the next key like this

```
country=DE
ap_scan=1

network={
	ssid="XXX"
	psk=""
	VhqO(J9MU:"
}
```

tuomashamalainen, 2020-12-16:
Thanks, this is a good clue. I just got the truncation problem here too testing with your original password, but it only happened after a couple of tries.

wpa_supplicant supports saving PSK as a hash and there's a command to create them from the SSID and original PSK, so what I think I will do is add this hashing. That way the stored PSK has only ASCII characters (which should prevent this problem) in it and we also don't store plain-text passwords.

Best,
Tuomas

jeLee6gi, 2020-12-16:
That sounds like the best solution to me, better than having to escape characters in user passwords. Thank you for the quick solution! In the meantime my new password should also do the trick.

closed 2022-05-17 by hifiberry