hifiberry/hifiberry-os #197: Optical source missing after 20201213 update

laurillev, 2020-12-19:
**Describe the bug**
Optical source is not visible anymore in Sources and no sound is produced from toslink input.

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
DAC+DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sources-tab in Hifiberry OS
2. Look through the list

**Expected behavior**
Expected to have Optical source as option and sound to be produced as before.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
DSP program is “DAC+ DSP Universal Version 12”

LarsenDX, 2020-12-21:
Hi,
Same here. Optical source missing since 20201213.

ndamkjaer, 2020-12-21:
Hi,
I also have this issue. 
Running Pi 4 with DAC+ DSP.

TheWaveHunter, 2020-12-21:
Hi,
Same here. Using optical input as primary source.
Pi 4, after update from 20201101 to 20201213

hifiberry, 2020-12-21:
Ok, this is a bug in the upgrade process. Try the following

rm /etc/hifiberry.state
/opt/hifiberry/bin/reconfigure-players

laurillev, 2020-12-21:
Unfortunately that didn't solve it for me.

Mine is a new installation, so it shouldn't have to do with upgrade. But thus I cannot temporarily go back to previous version either.

hifiberry, 2020-12-21:
What's the output of the reconfigure-players command?

laurillev, 2020-12-21:
cat: /tmp/eeprom.txt: No such file or directory
cat: /tmp/eeprom.txt: No such file or directory
/etc/hifiberry ~
~
pi
bluetooth
pi3orlater
arm7
dsp
localui
DAC+ DSP 1
Nothing changed, exiting...

hifiberry, 2020-12-21:
Hmm, this looks different than what it should be. What's the output of

grep toslink /opt/hifiberry/bin/reconfigure-players

laurillev, 2020-12-21:
grep toslink /opt/hifiberry/bin/reconfigure-players 
     FEATURES="$FEATURES spdifnoclock toslink"
   FEATURES="$FEATURES toslink"
      FEATURES="$FEATURES dsp toslink"

hifiberry, 2020-12-21:
Ok, that also looks right. I need to have another look.

laurillev, 2020-12-21:
Maybe that toslink "feature" is not correctly set for DAC+DSP somewhere.

hifiberry, 2020-12-21:
Yes, that's right. Question is just WHY

hifiberry, 2020-12-21:
Ok, looks like I found the problem. We'll fix it in the next release. For now, just use a workaround:

touch/etc/hifiberry/toslink.feature
/opt/hifiberry/bin/reconfigure-players

schwachstelle, 2020-12-23:
Looks like a typo; in which directory should the feature file be placed?

laurillev, 2020-12-23:
@schwachstelle use space between touch-command and '/etc/...'

schwachstelle, 2020-12-23:
@laurillev yes, but there is no directory /etc/hifiberry. Maybe /opt/hifiberry/etc ?

hifiberry, 2020-12-23:
Just create the directory

zaneclaes, 2021-01-04:
Just confirming that the following two commands, alone, worked for me with my Beocreate (note there was a typo with a slash after `touch` in the original):

```
touch etc/hifiberry/toslink.feature
/opt/hifiberry/bin/reconfigure-players
```

David91919, 2021-01-18:
I have the same issue and needed to make a fresh install so I can´t go back to the old version.

What are the exact commands for the fix ?

Which directory do I need to create?

Running these commands just gets me errors:
touch etc/hifiberry/toslink.feature          > touch: cannot touch 'etc/hifiberry/toslink.feature': No such file or directory

/opt/hifiberry/bin/reconfigure-players
```
# /opt/hifiberry/bin/reconfigure-players
cat: /tmp/eeprom.txt: No such file or directory
cat: /tmp/eeprom.txt: No such file or directory
/etc/hifiberry ~
~
pi
bluetooth
pi3orlater
arm7
dsp
localui
Beocreate 4-Channel Amplifier 1
Nothing changed, exiting...
```

hifiberry, 2021-01-18:
See https://github.com/hifiberry/hifiberry-os/issues/197#issuecomment-748946792

David91919, 2021-01-18:
> See [#197 (comment)](https://github.com/hifiberry/hifiberry-os/issues/197#issuecomment-748946792)

its not working :-( fresh install after download from website, root login

```
# touch/etc/hifiberry/toslink.feature
-bash: touch/etc/hifiberry/toslink.feature: No such file or directory
# /opt/hifiberry/bin/reconfigure-players
cat: /tmp/eeprom.txt: No such file or directory
cat: /tmp/eeprom.txt: No such file or directory
/etc/hifiberry ~
~
pi
bluetooth
pi3orlater
arm7
dsp
localui
Beocreate 4-Channel Amplifier 1
Nothing changed, exiting...
```

hifiberry, 2021-01-18:
Just put a space between touch and the filename

philaub, 2022-03-29:
> Ok, looks like I found the problem. We'll fix it in the next release. For now, just use a workaround:
> 
> touch/etc/hifiberry/toslink.feature /opt/hifiberry/bin/reconfigure-players

Are there any updates? The "next release" batch seems a bit odd now I think.