hifiberry/hifiberry-os #198: Overlapping partitions after upgrade

brannkobre, 2020-12-22:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Overlapping partitions after upgrade. I had issues with Spotify connection and I decide to allow wifi. When insert the SD into a laptop and I opened Disks (ubuntu disk management tool) I got  a message "can't have overlapping partitions"

**HiFiBerryOS version**
hifiberryos-20201101-pi4

**HiFiBerry sound card**
HiFiBerry DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2020-12-22:
We haven't seen this here and an upgrade should not change partitioning at all. What SD card size are you using?
What's the output of
fdisk -l

brannkobre, 2020-12-22:
Unfortunately I format it before I decide to sent you a  bug report. Next
time if it happens I will take more evidence.

hifiberry, 2021-01-20:
I close this now. Just come back if you see this again.

closed 2021-01-20 by hifiberry