hifiberry/hifiberry-os #199: Serial Port cannot be activated/Does not work

lukas-hllr, 2020-12-23:
**Describe the bug**
I'm not able to activate the Serial Port in the 'Interact' menu. Îf I try send to send a serial message on Startup anyways, I'm not able to receive anything. Even if I assume that I used the wrong baudrate (9600?) to receive the message, I can't see the Rx/Receive LED of the Arduino flashing. How do I activate the serial port corretly?

**HiFiBerryOS version**
hifiberryos-20201213-pi3

**HiFiBerry sound card**
PiFi DAC+ v2.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'INTERACT'
2. Activate interactions
3. Turn on the Serial Port (Nothing happens if I click on 'Serial Port Off', to activate the Port. Bug?)
3. Create any new interaction that sends data via the Serial Port (e.g. on Startup -> send 'Test')
4. Try to recieve the message with an Arduino

**Expected behavior**
I should be able to enable the Serial Port via the Webinterface.
The Arduino should recieve a Serial Message/The Rx/Receive LED should flash.

**Screenshots**
[image]
[image]

**Browser (if applicable)**
 - OS: Windows 10
 - Browser: Microsoft Edge (Chromium)
 - Version: 87.0.664.66

hifiberry, 2020-12-24:
Looks like you're not using a HiFiBerry sound card. Please first install a genuine HiFiBerry sound card.

closed 2020-12-24 by hifiberry

dodepier, 2021-01-18:
I have a hifiberry dac+ dcp on a raspberry pi 3 and can also not active the serial port. Or is the hifiberry dac+ dcp not supported?

hifiberry, 2021-01-20:
Is it a Pi3B or Pi3 B+? On the 3B+, the serial port is used for Bluetooth

dodepier, 2021-01-20:
It is a Pi3A+

hifiberry, 2021-01-20:
Same thing. You would need an USB-serial adapter for this