hifiberry/hifiberry-os #201: bugs in mobile webui

icstyle, 2020-12-27:
**Describe the bug**
On mobile web UI, I can't select other frequency squares in Sound Design. I can only modify the first one (which gets automatically selected). Also in listening modes, I can't select other modes.

**HiFiBerryOS version**
20201213 (problem exists since several months)

**HiFiBerry sound card**
DAC+DSP @ RPi4

**Browser (if applicable)**
 - OS: Android 11
 - Browser: Chrome
 - Version: latest

tuomashamalainen, 2021-01-14:
Hello and thanks for reporting this. I am able to reproduce this in desktop version of Microsoft Edge with mobile simulation, so I will investigate further.

Sorry for the inconvenience.

Best,
Tuomas from Bang & Olufsen

hifiberry, 2021-01-15:
We hope this can be fixed in the next release.

closed 2021-03-19 by hifiberry