hifiberry/hifiberry-os #204: When Analog Input is enabled but not active/playing, loud periodic noise output on speakers (Amp60)

tobiv, 2020-12-30:
**Describe the bug**
A very loud, scratch-like noise is being output when the Analog Input is enabled, but not active/playing. The noise is not continuous, but goes on for ~1 second, then off for ~1 second, and so on. It is so loud that it's audible even when Volume is all the way down at zero.

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
1. DAC+ ADC Pro
2. Amp60

**To Reproduce**
Steps to reproduce the behavior:
0. Analog Input disabled.
1. Select and play Source other than the Analog Input. It doesn't matter if sound is playing, as long as the Source is shown in Now Playing.
2. Go to Sources, enable Analog Input. Noise immediately starts, over sound from current Source or just by itself.
3. Go to Now Playing, and Stop/Pause current source. 
4. Now Playing switches to Analog Input and the noise stops.

(Setting "Start when Signal detected" makes no difference to this.)

**Expected behavior**
No scary noise by simply having Analog Input enabled.

**Screenshots**
N/A

**Browser (if applicable)**
 - OS: macOS 11.1
 - Browser: Firefox
 - Version: 84.0.1

**Additional context**
N/A

tobiv, 2021-01-03:
> **Edit to add:** LED D2 flashes in same interval as noise.

I also posted this on the hifiberry hardware support forum, suspecting a hardware issue due to the flashing LED.

hifiberry, 2021-01-04:
You might have set the sensitivity too high. Try changing the sensitivity.

tobiv, 2021-01-04:
I have tried all three sensitivities and also with "Start when Signal detected" turned off, that didn't help.

tobiv, 2021-01-10:
Further testing: Input is working fine if it's the first thing playing after a reboot. So the issue only after switching to another source.

treuwark, 2022-03-02:
The same here.

HifiBerry sound card
- DAC+ADC Pro
- AAMP60
Latest hifiberry os 28.02.2022

There is a loud scratch noise while enabling analog input. It doesn´t matter if aamp60 is used or not. 
Often analog input starts automatically playing although it isn´t set in the options "Start when Signal Detected". 

what´s wrong ?

hifiberry, 2022-03-03:
Try changing the sensitivity of the input.

treuwark, 2022-03-03:
Do you mean this ?

[image]

hifiberry, 2022-03-03:
yes

treuwark, 2022-03-04:
The setting has no effect. There is a scratch noise.
What to do ?

hifiberry, 2022-03-04:
I can't say more. As a workaround, disable analog input if you don't use it.

tobiv, 2022-03-04:
FWIW, if you stop the current source before turning on the input, it works. There is still a faint noise when turning on the input, but acceptable.

treuwark, 2022-03-05:
@hifiberry 
Disabling is no option. I have choosen this, because of the analog input.

@tobiv 
Thank you. 
In parts i can retrace that behavior you described. My problem as mentioned in my first post, analog input starts playing, although  "Start When signal Detected" is disabled.
Stopping in the web interface of a browser connected via lan, has no effect. In the ui icon is shortly changing. Other sources work correct for stop/start.

Where can stop/start be affected ?
Which log file should be checked ?
How to stop/start via command line for testing purpose ?

treuwark, 2022-03-05:
This is the output when stopping

Mar 05 19:07:17 hifiberry systemd[1]: Starting Store volume...
Mar 05 19:07:17 hifiberry systemd[1]: Starting Cleanup of Temporary Directories...
Mar 05 19:07:17 hifiberry systemd-tmpfiles[1251]: /usr/lib/tmpfiles.d/samba.conf:1: Line references path below legacy directory /var/run/, updating /var/run/samba → /run/samba; please update the tmpfiles.d/ drop-in file accordingly.
Mar 05 19:07:17 hifiberry store-volume[1250]: Storing ALSA volume...
Mar 05 19:07:17 hifiberry systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 05 19:07:17 hifiberry systemd[1]: Finished Cleanup of Temporary Directories.
Mar 05 19:07:17 hifiberry systemd[1]: store-volume.service: Deactivated successfully.
Mar 05 19:07:17 hifiberry systemd[1]: Finished Store volume.
Mar 05 19:08:06 hifiberry python3[388]: INFO: controller - sent Pause to org.mpris.MediaPlayer2.alsaloop
Mar 05 19:08:06 hifiberry python3[388]: 127.0.0.1 - - [05/Mar/2022 19:08:06] "POST /api/player/playpause HTTP/1.1" 200 2
Mar 05 19:08:06 hifiberry beocreate2[723]: Source 'alsaloop' has deactivated.
Mar 05 19:08:06 hifiberry python3[388]: INFO: http_post - posted metadata update to http://127.0.0.1:80/sources/metadata ({'artist': None, 'title': None, 'albumArtist': None, 'albumTitle': None, 'artUrl': None, 'externalArtUrl': None, 'discNumber': None, 'tracknumber': None, 'playerName': 'alsaloop', 'playerState': 'stopped', 'streamUrl': dbus.String('alsaloop://', variant_level=1), 'playCount': None, 'mbid': None, 'artistmbid': None, 'albummbid': None, 'loved': None, 'wiki': None, 'loveSupported': False, 'tags': [], 'skipped': False, 'host_uuid': None, 'releaseDate': None, 'trackid': None, 'hifiberry_cover_found': False, 'duration': 0, 'time': 0, 'position': 0, 'positionupdate': 1646503686.8503897, 'trackId': None})
Mar 05 19:08:06 hifiberry python3[1028]: INFO: root - playback status changed from stopped to playing
Mar 05 19:08:07 hifiberry python3[388]: WARNING: controller - Metadata without artist, album or title - what's wrong here? [alsaloop, playing] None: None (None) None
Mar 05 19:08:07 hifiberry beocreate2[723]: Source 'alsaloop' has activated.
Mar 05 19:08:07 hifiberry python3[388]: INFO: http_post - posted metadata update to http://127.0.0.1:80/sources/metadata ({'artist': None, 'title': None, 'albumArtist': None, 'albumTitle': None, 'artUrl': None, 'externalArtUrl': None, 'discNumber': None, 'tracknumber': None, 'playerName': 'alsaloop', 'playerState': 'playing', 'streamUrl': dbus.String('alsaloop://', variant_level=1), 'playCount': None, 'mbid': None, 'artistmbid': None, 'albummbid': None, 'loved': None, 'wiki': None, 'loveSupported': False, 'tags': [], 'skipped': False, 'host_uuid': None, 'releaseDate': None, 'trackid': None, 'hifiberry_cover_found': False, 'duration': 0, 'time': 0, 'position': 0, 'positionupdate': 1646503687.9065309, 'trackId': None})
Mar 05 19:08:10 hifiberry beocreate2[723]: Checking MPD database update status...

tobiv, 2024-01-17:
Yes, it is still happening, just tested (v20230404). Should I try with the latest alpha?

hifiberry, 2024-01-17:
Feel free to try. Nothing specific has been done to address this, but as you seem to be the only user with this issue, it might just work.

treuwark, 2024-01-19:
> Feel free to try. Nothing specific has been done to address this, but as you seem to be the only user with this issue, it might just work.

No, its still active. I didn´t follow up at that time, because there was no improvement.