hifiberry/hifiberry-os #205: An issue with the volume slider.

skumka, 2021-01-01:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
I experience an issue with the volume slider.
It works on a laptop via a web browser, it does not work on a dedicated 7" raspberry touchscreen.

**HiFiBerryOS version**
the latest one from December 2020

**HiFiBerry sound card**
DAC2 HD+DSP".

**Expected behavior**
Volume slider on dedicated touchscreen should work like via web browser.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
HiFiBerry with dedicated 7" raspberry touchscreen

skumka, 2021-01-14:
Am I the only one seeing this issue?

hifiberry, 2021-01-20:
There are very few users with a touchscreen. We see if we can reproduce this.

eshcheglov, 2021-01-24:
One more user with touchscreen is here. Yes, I can confirm this bug. 

Slider works correctly in web interface, but reaction to touchscreen events is unpredictable. Sometimes it works, but only once. Sometimes slider jump to desired position, but then immediately jumps back to previous one. Sometimes it doesn't react to the touch at all. 

I can collect any debug information that you require. 

Version is 20201101, rPI4 + DAC2 HD+

stutteringp0et, 2021-02-07:
Many touchscreen issues with the december version.  Can't switch between all sound profiles, I can only pick between 2 (of 4).  When a song is playing I can't skip or start a new song from the music player.  it forgot the wifi access point I used while running the upgrade.....all sorts of things aren't right anymore.  I'm going back to the November version when everything worked except switching back from a bluetooth connection.

ajmccaus, 2021-04-09:
I can confirm the same, no volume control and inconsistent 'now playing' updates on the 7" official touchscreen for the Digi+ and Amp2 HATs with RPi 4.

closed 2024-01-12 by hifiberry