hifiberry/hifiberry-os #208: Support USB-in as an audio source

carderne, 2021-01-03:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Add support for USB audio input (on the Raspberry Pi's USB ports) as an audio source.

**Describe alternatives you've considered**
Buy a DAC/ADC.

**Additional context**
Is this desirable, would it require new code or just shell/config, and can I help? Don't know much C++ and not very familiar with Buildroot but would be happy to tinker away if nobody else does it.

(I want this so that I can use sources like a USB turntable or USB microphone with my Raspberry Pi+HifiBerry Amp2.)

Original forum question [here](https://support.hifiberry.com/hc/en-us/community/posts/360014333177-USB-audio-input-on-HifiBerryOS).

hifiberry, 2021-01-03:
This would require quite a lot of code as an external USB input runs asynchronously to the DAC clocks. Therefore, it would require some sort of sample rate conversion. If you want to look into this, you might have a look at the alsaloop code and add sample ratze conversion:
https://github.com/hifiberry/alsaloop

carderne, 2021-01-04:
Thanks for the quick response, I'll have a look into it but sounds like I'll be buying the DAC+ ADC instead. :+1:

closed 2022-07-21 by hifiberry