hifiberry/hifiberry-os #209: HiFiBerryOS not connect to NAS

armatron, 2021-01-04:
**Describe the bug**
When trying to connect my NAS drive (My Book World Edition) in HiFiBerryOS I get message: 
_Destination unavailable. Available shares on the selected NAS could not be listed._

**HiFiBerryOS version**
20201202

**HiFiBerry sound card**
AMP2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to General and then Music Player Daemon
2. Click on Add Other Server
3. Type IP of NAS and then User and Password
4. See error

**Expected behavior**
NAS should connect and I should be able to see artists in Music page.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
I also manually created /etc/smbmounts.conf and run /opt/hifiberry/bin/mount-smb.sh. This procedure was successful in v20200629 but now I got message:

_mount -t cifs -o user=,password=,rw /data/library/music/Duatron,//192.168.1.2/Public/Muzika,hifiberry,*****,vers=1.0
mount: /data/library/music/Duatron,//192.168.1.2/Public/Muzika,hifiberry,*****,vers=1.0: can't find in /etc/fstab._

hifiberry, 2021-01-11:
Can you post the content of your smbmounts.conf file (replace the password by xxx)

TonyTromp, 2021-03-25:
Found an issue with the mount-smb.sh script, which causes NFS shares which have spaces, to fail to mount.

I have modded the script on my system replacing the greedy grep from /opt/hifiberry/bin/mount-smb.sh
#/opt/hifiberry/bin/mount-smb.sh
....
while read -r line; do
  # Split the line first
  readarray -d \; -t parts <<< "$line"
  ...
done < /etc/smbmounts.conf

hifiberry, 2021-03-25:
@TonyTromp Can you send a pull request to the development branch?

TonyTromp, 2021-03-29:
Done. Its in Bugfix/209 #252

TonyTromp, 2021-03-30:
Wait with the pull as i also saw a related issue with /opt/beocreate/beo-extensions/mpd/index.js creating folders 'undefined-My Share' <- Will make a fix for both Undefined variable and the whitespace issue in the mount folder (destionation).

TonyTromp, 2021-03-30:
All solved now in pull request #253 . I have tested this change in different conditions (eg. clean system vs already having this issue) and all is good now. Additional code has been added so changes in Beocreate ui is not longer needed. Please check and merge.

TonyTromp, 2021-03-30:
All solved now in pull request https://github.com/hifiberry/hifiberry-os/pull/254 . I have tested this change in different conditions (eg. clean system vs already having this issue) and all is good now. Additional code has been added so changes in Beocreate ui is not longer needed. Please check and merge.

closed 2024-01-12 by hifiberry