hifiberry/hifiberry-os #211: now playing screen on start-up for Hifiberry OS hdmi

installaudio, 2021-01-06:
**Is your feature request related to a problem? Please describe.**
When restarting my hifiberry Dac I need to plug in a mouse each time to click the button in lower left corner to show the now playing screen. 

**Describe the solution you'd like**
On first start up can the now playing screen be showing - users can then click this down if they want to access settings / admin via hdmi.

**Describe alternatives you've considered**
no alternatives known

**Additional context**
I use a hdmi display plugged into a pi4 with DAC HD - the display shows what is playing in the room

hifiberry, 2021-01-15:
This should be the normal behaviour, but the browser seems to behave a bit strange in the last release. I hope we can fix this.

installaudio, 2021-01-20:
yes please - bug fix in new version hopefully

hifiberry, 2021-01-20:
Problem is that it's going well on a "normal" browser and the local browser doesn't have a full javascript console which makes it really hard to debug. It might not be fixed in the next release.

hifiberry, 2021-07-22:
This has been fixed in the May 21 release

closed 2021-07-22 by hifiberry