hifiberry/hifiberry-os #212: Spotify artwork from LMS not showing on Hifiberry OS

installaudio, 2021-01-06:
**Describe the bug**
Our system use LMS with spotify & tidal integrated via a plug in, Spotify & tidal artwork is not showing up in the Hifiberry OS now playing screen / HDMI displays. It keeps the artwork from previous track played (which is stored on local NAS). Spotify & Tidal artwork is appearing well, without issue, on LMS and on iPeng iOS (control app)

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
DAC HD

**To Reproduce**
Steps to reproduce the behaviour:
play Spotify / tidal stream via LMS / squeezelite source 

**Expected behavior**
Expect to see artwork from spotify and tidal on Hifiberry OS

**Screenshots**
advise if required

hifiberry, 2021-01-06:
Is just the artwork incorrect or also the song information?

installaudio, 2021-01-06:
Sorry I was unclear, 
Its just the artwork that is wrong, the song information is correct.

hifiberry, 2021-01-06:
In this case, it's not a bug, but something that simply isn't implemented. Our Squeezelite controller expects local artwork files, that are not there with streaming services.

installaudio, 2021-01-06:
Some further information,
we have noticed that within spotify - albums that are added to our favourites onlnie appear with the LMS stored music, and these albums (which stream from online) do not have artwork stored locally but they do display artwork correctly when played via LMS 'My Music'.